motioneye-project/motioneyeos #2667: Message throttlewatch

Ferrox12, 2021-02-09:
Hi,
every minute MotioneyeOs write in the file "messages.log" this string "user.notice throttlewatch: currently: ok, temperature: 33 C".
Do I have to worry?

starbasessd, 2021-02-09:
Nope, it means it's not overheating, the temperature of the CPU is 33`C,
very nice.
Many times, PIs are stressed, put into bad airflow cases and overheat.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Ferrox12, 2021-02-09:
OK but is there a method to limit this message?
Instead of every minute once every 10?

starbasessd, 2021-02-09:
Which version of motionEyeOS are you using, and which Pi?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Ferrox12, 2021-02-09:
Raspberry pi 3 b+ 
motionEyeOS 20200606

starbasessd, 2021-02-09:
The file in question setting up throttlewatcher is
/etc/init.d/S11throttlewatch
Can it be edited?
You can
mkdir /data/etc/init.d
cp /etc/init.d/S11throttlewatch /data/etc/init.d/
nano /data/etc/init.d/S11throttlewatch
Change CHECK_INTERVAL=60 to 600 (10 minutes)
[ctrl]x, y, [enter] to save
reboot
Should change it to 10 minutes

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Ferrox12, 2021-02-09:
Thank you very much!!!

closed 2021-02-09 by Ferrox12

starbasessd, 2022-02-14:
New instructions in #2883