motioneye-project/motioneyeos #2668: Can't boot from USB on Pi4 and configure screens greyed out in hub

d4m4n, 2021-02-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026 on Pi4 2 gig hub no camera; 20200606 on Pi Zero cams 
### Board Model

Pi4 2 gig for the hub; 4) Pi Zero W with cams

### Camera

MMAL Pi cam v2

### Network Connection

Wired ethernet Pi4 hub; wireless G Pi Zeros

Hello all, I'm new to the whole raspberry pi world, but not computers. So I got bit by the pi bug and bought a Pi4 2 gig to use with 4) Pi Zero W cameras using motioneyeos after I saw a video on how easy it was to build on a youtube channel. So I have my other RP4 2 gig booting from USB which runs my OMV5. No problems there, was a little tricky figuring out the folder shares but I eventually figured it all out. I wanted this new pi to boot from USB also. When I turned the Pi on it wouldn't boot from USB, I tried replacing the 2 files in the root of eyeos on the SD card, I did the sudo apt update, sudo apt upgrade and made sure the bootloader was updated as well. Jumped around, different versions. tried different tricks and fixes.  Nothing worked. Finally, I reimaged an SDcard, put it in the Pi in disgust and it worked. It's been working ever since. I setup the cameras individually, went thru each IP, set res, got storage setup, assigned an IP address manually (reserved in router), no problems there. When I goto the hub IP address, the config on the left side is ghosted out. I have four cameras to choose from, but the settings are ghosted out. When I was running 2 cameras and a Pi Zero as a hub, I could change the settings remotely for each camera. That option is gone now. Did I do something wrong? Other than that, motoineyeos is THE ticket and the price is right and the cameras are pretty inexpensive. I even got the cameras working on my rokus using each individual cameras ipstream:port. 

Thx in advance for all your help, I really find these forums helpful, I don't know where'd I'd be without them.

d4m4n, 2021-02-10:
Sorry, I should've added a screenshot. 

[image]

starbasessd, 2021-02-10:
If you need USB Boot, you can switch to motion/motionEye on RaspberryPiOS,
or Ubuntu 20.04 or another flavor of Linux OS that supports USB Boot.
Usually, when the greyd out options occurs, you've previously set up a
password, and clicked "Remember Me' in the log on.
To fix that, clear cache and cookies in your browser, and log in normally
to 'admin'.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

d4m4n, 2021-02-10:
Thanks for your swift reply. USB boot isn't a big priority. I just made a post about it so the author is aware. And your suggestion worked, problem fixed!!!

starbasessd, 2021-02-10:
It is a known issue, and something I've been wishing for since I came
across the project several years ago...
Especially with my testing support and constantly re-imaging SDCards, I go
through quite a few, vs all my other Pi projects where I can boot to USB.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

d4m4n, 2021-02-10:
So you're saying I can run Raspberry Pi graphical OS and motioneyeos as a hub? I should probably do that, since I've already got another USB thumb drive configured for SSH and VNC, and was gonna buy ANOTHER Pi to tinker with since the other 6 I have are doing jobs!

starbasessd, 2021-02-10:
You can run RaspberryPiOS Desktop or Full, and install motion/motionEye on
that.
Follow these instructions:
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
due to issues getting python2.7 and pip2 to install correctly.
It will also allow installation on Raspberry Pi OS Lite for headless
operation (like motionEyeOS)
The Docker version works well for me too, if you are into that sort of
thing. If needed, I use the
sudo apt install docker.io
and the instructions from here:
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker

Let me know if you need further help...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

d4m4n, 2021-02-10:
Will do, thanks so much. I will probably tackle the installation tomorrow. Slightly getting off topic, I tried installing plex media server using docker in OMV5 and I couldn't get it to work. Some of the files and commands disappeared off omvextras.org and the youtube video was out of date I guess. It was VERY confusing.  I gave up and just did a clean fresh install of OMV5 and connected a network drive to my PC for plex.  I'll tackle dockers at another time LOL I don't want to screw anything up. 

I just had another thought, would motioneye work on a dedicated pi running OMV5 with docker installed?

starbasessd, 2021-02-10:
You must take into consideration PIs are SBCs, not PCs with limited
resources and limited CPUs. The more you request of it, the less its
performance/responsiveness, especially when doing things with images/videos.
Can it be done? Sure. I used to run WinXP in a VM in Ubuntu VM in Win7
Host, but none were happy.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

d4m4n, 2021-02-10:
Gotcha. I've made my way to step 8 in the installation procedure, no problems so far, but I'm lost here:

Add an init script, configure it to run at startup and start the motionEye server:

I'm currently SSH'd and VNC'd in to my Pi OS.

starbasessd, 2021-02-10:
If you follow the steps, it automatically does that for you.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

d4m4n, 2021-02-10:
OK, I did that, the wording was a little confusing, I thought I had to make something like a .bat script for windows startup. I rebooted. I'm not seeing anything.

starbasessd, 2021-02-10:
Are you going to http://[ip_address_of_Pi]:8765 fron another computer web
browser, or http://localhost:8765 on the Pi Desktop Browser?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

d4m4n, 2021-02-10:
port 8765 did the trick :)

starbasessd, 2021-02-10:
Enjoy.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-02-10 by starbasessd