motioneye-project/motioneyeos #2669: Action buttons with PCA9685 board - servos do not move.

jojo2203, 2021-02-12:
Hello,

apart from importing the files and making them executable, do i also have to import the phyton environment under motioneyeos?

I use a Raspberry pi 3B with motioneyeos 20200606, the Raspberry pi camera v1.3 and the board PCA9685 with 2 Servo SG90. I think I have wired everything correctly. Ground, SCL,SDA, 3.3 V, 5V and EO > GPIO18 are connected.

I followed the instructions from https://github.com/luetzel/motion_eye_servo_action.

I can see the buttons, can also operate them, the servos work, I have checked this with an external servo tester.

Unfortunately, the servos do not move. Ideas?
Thanks.

lanche11, 2021-02-15:
Ditch the PCA9685 It won't work on MEOS. Use the code below. Remember to "chmod"all files *_1 created    
This method creates " WARNING: up_1: command has finished with non-zero exit status: -15" each cycle in motioneye.log
The servo jerking does not appear because of the termination. Use 38 & 40 gpio pins. Works on PiZero & Pi3B & B+ ME & MEOS. Does not work on Pi4 or Tinkerboard MEOS. Tinkerboard Debian needs ASUS.GPIO as GPIO

right_1

#!/usr/bin/env python
import RPi.GPIO as GPIO
import time
import os

GPIO.setmode(GPIO.BOARD)
GPIO.setwarnings(False)

myservo = 38  # Use 40 PWM pin for up and down

GPIO.setup(myservo,GPIO.OUT)
pwm = GPIO.PWM(myservo,60)
pwm.start(1) #change to (15) for left

while True:
                print "right"
                pwm.ChangeDutyCycle(1)#change to (15) for left and can be anything between 1 & 15
                time.sleep(.02)  # increase or dercease for speed of sweep (.08)fast (.01)slow
                os.system('kill $PPID')

jojo2203, 2021-02-17:
thanks for your help, i have implemented it. entered the code in right_1 and made it executable. and the other files up_1, left_1 and down_1 too. connected one servo to pin 2 (+), pin 6 (ground) and pin 38 (signal) for testing, but elider without success. do you have any other ideas and tips?

thanks. jojo

lanche11, 2021-02-18:
What error appears if you are in directory  /data/etc  and execute ./right_1    ? 
You should get movement  and
[root@meye-ba27dd12 etc]# ./right_1
right
Terminated
[root@meye-ba27dd12 etc]#

jojo2203, 2021-02-18:
Hello lanche11,

please see the screenshot.
[image]

lanche11, 2021-02-18:
Looks normal. Did you have movement? Is your 38 Board or BCM 20

lanche11, 2021-02-18:
Line
pwm.ChangeDutyCycle(5)
needs to be 
pwm.ChangeDutyCycle(1)  same as   pwm.start(1)

jojo2203, 2021-02-21:
i got it working now, thanks again for the help. i still had a contact problem in the experimental setup. i changed the 60 to 20 to reduce the step size. it works very reliably when panning up and down. when panning from left to right, it goes through completely twice, then it hangs. i'll probably have to play with the parameters and limits a bit more.

thanks jojo