motioneye-project/motioneyeos #2670: Freeze OrangePi

thalesmaoa, 2021-02-15:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: Motion 4.2.2+gitUNKNOWN.

### Board Model

I am using the following board/model: Orange Pi One.

### Camera

I am using the following type of camera: Cheap Chinese Brand (V380 - Compatible).

My camera model is: Generic.

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet Cable.

### Peripherals

I am using the following peripherals that I consider relevant to this issue: None

### Log Files

I consider the following log files relevant to this issue:

motioneye.log
messages.log
 
Hi there, I never faced a single problem since I start using MotionEye OS 3 years ago. However, I'm facing it now. Since I'm using an Orange Pi, I was convinced that it was the problem. However, after looking the logs, I'm not so sure.

1) Recently my local network provider changed my modem and they obligated me to have IPv4 + IPv6. I can't disable IPV6. MotionEyeOS worked without a problem since I manually set IPv4. However, after another freeze and looking into messages.log, I found the following:

```
Feb 15 03:19:53 motioneyeOS user.err motion: [3:ml3:Porta] [ERR] [ALL] motion_init: Error capturing first image
Feb 15 04:18:12 motioneyeOS user.notice network: eth0 has no IP address
Dec 31 21:00:08 motioneyeOS daemon.err rpcbind[1211]: cannot create socket for udp6
Dec 31 21:00:08 motioneyeOS daemon.err rpcbind[1211]: cannot create socket for tcp6
Dec 31 21:00:08 motioneyeOS user.info sntp[1340]: sntp 4.2.8p13@1.3847-o Wed Nov  6 17:33:54 UTC 2019 (1)
Dec 31 21:00:09 motioneyeOS user.info sntp[1340]: 1969-12-31 21:00:03.078170 (+0300) +1613396224.529016 +/- 1075597483.054597 0.pool.ntp.org 198.27.76.102 s2 no-leap
```

Am I wrong or it just lost its IP? It is also trying to get an IPv6? How do I completely disable it?

2) A also found another interesting log inside motioneye.log

```
2021-02-15 10:40:27: [motioneye]     INFO: downloading firmware version 20200606
2021-02-15 10:40:35: [motioneye]     INFO: extracting firmware
2021-02-15 10:41:05: [motioneye]    ERROR: Exception in callback <functools.partial object at 0xb40a6330>
Traceback (most recent call last):
  File "usr/lib/python2.7/site-packages/tornado/ioloop.py", line 758, in _run_callback
  File "usr/lib/python2.7/site-packages/tornado/stack_context.py", line 300, in null_wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/platformupdate.py", line 54, in perform_update
Exception: firmware extracting failed
```

It is not updating? Is there a way to fix it?

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5982029/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5982035/messages.log)

starbasessd, 2021-02-15:
Which version of motionEyeOS? You posted the version of motion, but not which image file you are using.

thalesmaoa, 2021-02-15:
Sorry @starbasessd, I edited my original post: 0.41.

starbasessd, 2021-02-15:
No, I am referring to which install image file:
motioneyeos-orangepione-20200606.img.xz
motioneyeos-orangepione-20200203.img.xz
motioneyeos-orangepione-20190911.img.xz
or 
motioneyeos-orangepione-dev20201026.img.xz

thalesmaoa, 2021-02-15:
I see, none. The #2157 I pointed my problem. I needed to compile it with the available version that time to use NFS. I suppose 20191021.

starbasessd, 2021-02-15:
If you are using another base OS, then you probably need to be asking over on the motionEye github.
https://github.com/ccrisan/motioneye/issues

thalesmaoa, 2021-02-15:
Should I reopen there and close here?

starbasessd, 2021-02-15:
Yes, please.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-02-15 by thalesmaoa