motioneye-project/motioneyeos #2671: SMTP Server Error with GMail

JohnnyManziel, 2021-02-15:
Hello,

When I try to set up the Motion Notification with Gmail I consistently get this error: "Notification email failed: check SMTP server name!"

Here is what I have tried to fix it:
-different SMTP port numbers: 25, 465 (w/o TLS), 587 (w/TLS)
-SMTP Account and Address with and without the @gmail.com at the end
-two factor authentication for the gmail account
-allow less safe apps turned on
-using a user generated password

Here is the error I get in my log:
1970-01-11 04:36:20: [sendmail]    ERROR: failed to send mail: [Errno -2] Name or service not known
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 225, in on_message
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 53, in send_mail
  File "usr/lib/python2.7/smtplib.py", line 256, in __init__
  File "usr/lib/python2.7/smtplib.py", line 317, in connect
  File "usr/lib/python2.7/smtplib.py", line 292, in _get_socket
  File "usr/lib/python2.7/socket.py", line 557, in create_connection
gaierror: [Errno -2] Name or service not known

Here are the settings I use in the Motion Notifications:
Send an Email: On
Email Address: address@gmail.com
SMTP Server: smtp.gmail.com
SMTP Port: 587
SMTP Account: address (no @gmail.com)
SMTP Password: password (app generated)
From Address: address@gmail.com
Use TLS: On
Attached Pictures Time Span: 30 seconds

starbasessd, 2021-02-15:
SMTP password needs to be the same as what you use to log into gmail with.
SMTPAccount is google.user@gmail.com
I send to google.user@gmail.com
My From is google.user@gmail.com
Use TLS ON
Click APPLY at the top BEFORE clicking Tesst Email.
[image: tempsnip.png]

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JohnnyManziel, 2021-02-15:
Thank you, my password for login to gmail and on motioneyeos were the same, and I ensured that I applied before testing.

starbasessd, 2021-02-15:
SMTP Account is WITH @gmail.com.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JohnnyManziel, 2021-02-15:
I have tried that with and without the @gmail.com and no luck either way, thanks.

starbasessd, 2021-02-15:
If you are using motionEye20200606 on any device (since you didn't answer any of the standard questions) please upgrade to dev20201026 or downgrade to 20190911 (Unless you are using a Pi4B, in which case just upgrade to 20201026)

starbasessd, 2021-02-18:
SMTP Account: address (no @gmail.com)
needs to be WITH @gmail.com

SMTP Password: password (app generated)
Needs to be the one you use in your web browser to access your account. motionEyeOS doesn't use the Google Gmail API.