motioneye-project/motioneyeos #2672: change keyboard language on motioneyeos

githlp, 2021-02-16:
### Preliminary Docs
confirm/confirm

running motionEyeOS version: 20200606 on Raspberry PI4

**Is there a possibility to change the keyboard layout on the login shell ?**

I found this within the closed issues
https://github.com/ccrisan/motioneyeos/issues/1668

but i think the answer is invalid, not the question.

Motioneyeos **IS** the only OS on that Raspi4 and i cannot change the keyboard language layout on my Raspi4 Hardware with a jumper or something.

The question is what config file must be changed to change the keyboard layout since typical tools as raspi-config or the adjustment of /etc/default/keyboard does not change anything..

starbasessd, 2021-02-16:
Does your prefered keyboard layout work when you SSH in?

githlp, 2021-02-16:
no, with ssh the remote [shell|keyboard] config is applied.

The question was about changing the keymap on the local login shell on the BuildRoot environment of latest MotionEyeOS.

starbasessd, 2021-02-16:
I realize that. Before I tag it, though, I wanted to see if there is a
viable work-around.
Have you turned on Debug Mode (to enable the filesystem to be read/write,
instead of read only, make your change, then turn off Debug Mode?
motionEyeOS was built as a minimalist, headless system, and motionEye is
for sitting on top of a full OS.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

githlp, 2021-02-16:
well i remounted root rw, otherwise i would not have been able to place a keyboard file in the new default folder but that file is not recognized, so i was wondering where language or layout values are set.

starbasessd, 2021-02-16:
It appears to be a ThingOS limitation. Several other packages using it seem
to have the same issue, even placing the 'right' files where they belong.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?