motioneye-project/motioneyeos #2673: Version 20200606 - SSH - connection refused

zekirma, 2021-02-16:
I try to add RSTP camera to the last version. The URL is invalid. So I try to connect vai SSH on the PIZero but the connection is refused. All the rest is working well.
I try to set de SSH as initlal startup by creating a empty file SSH in the boot partition, but it does not change any things.

So how enable the SSH on my Pizero with this last verstion og Motioneyeos?

starbasessd, 2021-02-16:
Are you able to connect to the WebGUI?
You can enable SSH under Expert Settings.
The username is "admin" [without the quotes]
password is blank unless you changed it in the WebGUI
On the PiZero, did you include the wpa_supplicant.conf file to connect to
wireless?
Can you connect a monitor and see what IP address it's getting?

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zekirma, 2021-02-16:
I'm in the GUI in the experts section, but I don't see any SSH on/off parameter. Is it Débug?

zekirma, 2021-02-16:
Ok the debug give me acces to putty on port 22. 

Now I have to resolve the RSTP invalid url message when I try to add camera

zekirma, 2021-02-16:
The SSH was activated by set on the debug mode

closed 2021-02-16 by zekirma

odiseo123, 2021-02-21:
Hi @zekirma 
I am having the same issues with the v20200606 on a RPi 3 B+. I have a Putty connection profile which has stopped working after the update with 'connection refused'. I can access the GUI and the option 'Enable SSH server' is ON under Services.
Could you confirm that by _debug_ you mean 'Enable Debugging' under 'Expert settings'? I have turned that ON and rebooted, but without luck.

I am wondering what else you may have tried.
Any help is appreciated.

Andres