motioneye-project/motioneyeos #2675: Is it possible to change the shutter speed on a schedule?

Ryan5374, 2021-02-17:
I am running on a pi zero with 20200606. Making the shutter speed slow is the only way that my pi camera can see at night, but it becomes completely white during the day. Is there a way to change it on a schedule?

starbasessd, 2021-02-17:
How are you changing the shutter speed?

Ryan5374, 2021-02-17:
At the moment, I just go to the video device setting on the pi that has the camera connected.

starbasessd, 2021-02-17:
Specifically?
Are you using Fast Netwok Camera?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Ryan5374, 2021-02-17:
Yes, it's a fast network camera added to a hub running on a tinkerboard