motioneye-project/motioneyeos #2676: Help Wanted: How to

Ru-Pirie, 2021-02-17:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 

- motionEyeOS 20200606

### Board Model

I am using the following board/model:

- 2 x Raspberry Pi Zero W
- 1 x Raspberry Pi 3 B+ as a hub

### Camera

I am using the following type of camera:

- 2 x MotionEyeOS Network Cameras

My camera model is:

- 2 x Night vision camera module for Raspberry Pi – 160° (Fisheye)

### Network Connection

My motionEyeOS unit is connected to the network via:

All of the cameras are running over WIFI.

### Peripherals

N/A

### Log Files

N/A

I'm wondering is it possible to use the main Pi 3 B+ to record the video feed of the two Pi Zeros. I have looked through the settings and I can't seem to find one that would allow me to record (motion triggered) the two cameras on the main Hub Pi. I can get the indervidual cameras recording but not the main Pi.

starbasessd, 2021-02-17:
On the Pi3 hub, add the cameras as Network Cameras or Simple MJpeg Cameras, unless you feel the need to manage them only on the Pi3 Hub.  If you are running the PiZeros as Fast Network Cameras, then you have to add as either Network Cameras or Simple MJpeg Cameras.
So, on the Pi3 Hub you'd put in http://[ip_address_1]:8081 User and password as needed, and pick one of the 3 types (Network, Simple MJpeg, or PiCam. For the second you do the same thing, but use http://[ip_address_2]:8081

Ru-Pirie, 2021-02-17:
Ok, Ill look into it.

Ru-Pirie, 2021-02-17:
I tried to add the cameras as MJPEG and all I got was a green screen which soon after connecting went grey.
[image]

Ru-Pirie, 2021-02-17:
This occured when trying to add the camera as a network camera

starbasessd, 2021-02-17:
How do you have he PiZeros set up?

Ru-Pirie, 2021-02-17:
In what way do you mean?

Ru-Pirie, 2021-02-17:
They are both connected to the same network and have video streaming enabled.
[image]

starbasessd, 2021-02-17:
Do you have the PiZers set up with Fast Network Camera turned on?
I would also suggest downgrading to 20190911 or upgrade to dev20201026.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Ru-Pirie, 2021-02-17:
I cannot set them up like that due to the fact that I want to be able to use motion detection on them.

Ru-Pirie, 2021-02-17:
Seems to be whats happening here: #2146

starbasessd, 2021-02-17:
Unless you are using an odroid and 20190911, I doubt it.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Ru-Pirie, 2021-02-17:
Ok, let me refraise,it is the same thing that is happening to me. I add it it goes green connecting, then grey and disconnected.

Ru-Pirie, 2021-02-18:
This issue has been resolved.

closed 2021-02-18 by Ru-Pirie

sts-ryan-holton, 2021-07-23:
@Ru-Pirie What was your fix? I've got two Pi's, one as a hub, and the other with a USB webcam with video streaming enabled. I need the main hub to accept the video feed from the other Pi and save video, if I choose the simple MJPEG camera on the hub and input the video streaming URL, it works, but I don't have the option to record, when I switch to network camera I use the same URL but I get a green box.

starbasessd, 2021-07-23:
Did you try remote motioneye camera?
Are you using motionEye or motionEyeOS? If mEOS, which version(s)?
What URL are you using?
Is Fast Network Camera involved?
You need to open a new Issue, answer the questions. If you want you can reference this issue, but the original poster has an ODroid, and a Pi. His issue wasn't necessarily caused by the same thing as yours.