motioneye-project/motioneyeos #2677: Is the project still alive?

ZeroDot1, 2021-02-18:
I see a lot of open issues here, and no changes for some time.

I hope that the project is still alive.

CC: @ccrisan

starbasessd, 2021-02-18:
Yes, the project is alive. Last dev image is getting 'long in the tooth',
but is still being worked on.
I am slowly working on cleaning up the issues, labeling for others to look
though, helping where I can, and debating on closing issues older than 2
years due to the large number of updates since then. 20190911 for the pre-
Pi4 and dev20201026 are both very useable and stable across the Pi
platforms, which I support. I make changes and suggestions in the issues
categories nearly every day.

Do you have a specific question for here or motionEye on another OS?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ZeroDot1, 2021-02-18:
OK, thanks for the quick reply, All my questions have been answered, I'll close this.

Thank you very much for your good work.

closed 2021-02-18 by ZeroDot1

ccrisan, 2021-02-18:
Yes, the project is still alive and I am still around. Nevertheless, the amount of issues is overwhelming and the spare time I have to spend on this project is very limited.

My full appreciation goes to @starbasessd who does a great job in sorting out issues and providing support for the community.