motioneye-project/motioneyeos #2678: rpi zero w wpa_suplicant

mchaos, 2021-02-18:
It's nice there is a script that does everything for me, however I am past that point and I just want to know where the wpa_sup file needs to go and what the full file name is.

Surprising that this info is not on the github

Thanks,

starbasessd, 2021-02-19:
wpa_supplicant.conf goes in the same place as in RaspberrryPiOS, the /boot partition, after the SDCCard has first been imaged, which is why it's not in the documentation, since motionEyeOS is usually not the first uses of a PiZeroW. I will endeavor to get that documented, though, thanks.

starbasessd, 2021-02-19:
BTW, it is clearly documented:
https://github.com/ccrisan/thingos/wiki/wpa_supplicant.conf#file-location-and-role
The Link is in the document for
https://github.com/ccrisan/thingos/wiki/wpa_supplicant.conf
Which is in the Wiki link for wpa_supplicant.conf in the right panel under Configuration...

closed 2021-02-19 by starbasessd

mchaos, 2021-02-19:
I figured it out after i posted it without the wiki. I couldn't find it for some reason. My bad. Any way I figured it out on my own and actually came back to close this.

Thanks,