motioneye-project/motioneyeos #2679: cant conect to wifi with motion eye (5GHz network)

enormouspenis69-max, 2021-02-24:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

enormouspenis69-max, 2021-02-24:
I am running motionEyeOS version: 0.42.1 

I am using the following board/model: Raspberry PI 3B v1.2

Camera
I am using the following type of camera: raspi camera v2.1

Network Connection
My motionEyeOS unit is connected to the network via: Ethernet but  wont connect to wifi 5ghz

i have followed all available tutorials on how to write the wpa_supplicant.conf file and it doesnt work. it works via ethernet fine. i used notepad++ with EOL Conversion Unix(LF) and saved it on the sd from windows. iv tried it multiple times (obviosly with my wifi credentials changed and country code is au) using this code:
country=US
        update_config=1
        ctrl_interface=/var/run/wpa_supplicant
        
        network={
            scan_ssid=1
            ssid="MyWiFiSSID"
            psk="S3cr3tp@$$w0rc|"
        }
perhaps the code needs to be differnt for a 5ghz network. any help will be appreciated:)

starbasessd, 2021-02-24:
The Raspberry Pi 3B V1.2 2015 only has WiFi 802.11 A/B/G/N (2.4GHz)
The Raspberry Pi 3B+2017 has WiFi 802.11 A/B/G/N/AC (2.4GHz + 5GHz)
THere are numerous websites with the WiFi specs per revision
https://www.raspberrypi.org/documentation/faqs/#networking-builtinwifi

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

enormouspenis69-max, 2021-02-24:
big help i couldnt figure that out thanks for the help,
 i need a dongle or upgrade my raspi.
legend,  cheers:)

closed 2021-02-24 by starbasessd