motioneye-project/motioneyeos #2680: MotioneyeOS Unable to connect to it unless it pings out first

zacharym8, 2021-02-24:
### Preliminary Docs
I confirm that I have read the contributing guide before opening this issue.
I confirm that I have read the FAQ before opening this issue.
And I have also dived through open and closed tickets seeing if there were any issues called out to be similar.

### motionEyeOS Version
I am running motionEyeOS version: **dev20201026**, though **I also tested 20201026 and 20190911** with the same exact issue presenting itself. I originally started with 20201026 and am now on the dev version hoping it had firmware updates.

### Board Model
I am using a **Raspberry Pi Zero W**.

### Camera
I am using the following type of camera: **MMAL Camera**.
My camera model is: **Raspberry PI Camera with IR** (Exact camera is "kuman Raspberry PI Camera Module 5MP 1080p OV5647 Sensor" from amazon). **Connected by the FPC** cable directly to the Zero.

### Network Connection
My motionEyeOS unit is connected to the network via: **WiFi with a static IP** assigned to the device by my router using the pi's MAC. The same issue occurs on a separate Pi running off of WiFi and DHCP instead of a static IP.

### Peripherals
There are no peripherals involved.

### Log Files
I don't actually know if the following log files are relevant to this issue:

 * (attach e.g. motioneye.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/6038590/motioneye.log)

 * (attach e.g. boot.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/6038589/boot.log)

 
### The Issue at Hand
So right off the bat, the initial loading of the OS, the adding of the ssh file and the wpa_supplicant into boot, and the first boot of the device all go smoothly with no issue and I can access the http://ipaddress site to play around with the settings and see through the camera clearly. I can ssh into the pi, and ping it from my main computer no issue.

The issue occurs once it reboots, on any boot that is not the first set up boot. I cannot access the pi through ssh, the web gui, or even ping it from my host machine. The device is fully connected to wifi once I plug it into a screen with HDMI, has its IP still assigned, it can ping or download anywhere I point it, but I cannot access it from my main computer.

After several days of troubleshooting and googling I found a similar issue mentioned with Raspberry pi firmware and they mentioned that they could reach the device if it pinged out to the computer first. So I pinged my main computer from the pi and now my main computer could see it, ping it, ssh, or access the motioneye web page. So I can access the pi from any device on my network, only if I have the pi ping out first.
With that narrowed down I did some more googling and found that this seemed to be an issue encountered in their github page back in 2018 as shown in the following link: (https://github.com/raspberrypi/linux/issues/2617)
It seems they fixed it by making some changes to the pis firmware that I don't quite understand but it seems to be the same issue.

So the reason I am here is I cannot seem to find anyway to update that firmware within the motioneyeos, and if I load up a recent version of raspbian and then load motioneye onto that OS it does not experience this problem. However I like the OS version better, so I was hoping to figure out why I am having this issue with motioneyeos and if there is hopefully a quick way to fix it. Otherwise I will just see about making some sort of script to just ping all addresses on my network periodically but I think there is probably a more elegant solution than that.

Any help would be appreciated.

starbasessd, 2021-02-24:
First thing I notice is in boot.log, first boot is getting 
# Interface wlan0 has IP address [the correct ip]/8
but the 2nd boot is getting
 * Configuring wireless network: [the correct ip]/24
 If the IP for the Pi is, say, 192.168.10.99 but the Gateway is 192.245.0.1,
it would work in the first case, but not in the 2nd.
Normally, the first 3 octets match (figuring the /24)
Something isn't quite right.

zacharym8, 2021-02-24:
> First thing I notice is in boot.log, first boot is getting
> # Interface wlan0 has IP address [the correct ip]/8
>
> but the 2nd boot is getting
>
>     * Configuring wireless network: [the correct [...]

I will get that corrected, though I can say that they are all on the subnet, the router being 10.0.0.1 and this being essentially 10.0.0.2
I haven't had a chance to really start splitting networks off as I havent learned that far into networking yet.

starbasessd, 2021-02-24:
It also affects the route, where one can't see the other by default
(non-existent route generated) but can after the other pings the first.
TCP/IP is kinda strange when using effectively different sub-nets
(10.x.x.x/8 vs 10.x.x.x/24 are different subnets)

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zacharym8, 2021-02-24:
Just to now confirm I switched back from the static IP settings to the DHCP settings and it is now back to /8. The same issue persists, and just to double check against another device, the pi zero w I set up with updated raspian and motioneye has the exact same IP structure now, so it looks like:
10.0.0.1 - Router
10.0.0.2/8 motioneyeos on pi zero w - still having this particular connectivity issue.
10.0.0.3/8 latest raspbian with motioneye installed on pi zero w - connects perfectly fine upon boot.

starbasessd, 2021-02-24:
but does the 10.0.0.3 survive reboot?
Can you try 20190911?
Did you change the admin password on .2?
If you did, try to clear cache and cookies after the reboot, and don't
select "Remember Me' until fully set up and after several reboots.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zacharym8, 2021-02-25:
Alrighty, so I grabbed a new micro sd, flashed 20190911 to it, put the ssh file and wpa_supplicant in the boot drive, and loaded it on into the pi zero w. First boot loaded up fine but it actually now suffers from the same issue as mentioned above but now on the first boot as well. Its ip is the same /8 and I have not changed any passwords or anything as this is a fresh boot. I have cleared cache and cookies multiple times when trying to get through to the page but nothing seems to get through until I go onto the pi and ping my main computer, after that it is back to working normally for the rest of its uptime.

[motioneye(1).log](https://github.com/ccrisan/motioneyeos/files/6039626/motioneye.1.log)
[motion(1).log](https://github.com/ccrisan/motioneyeos/files/6039627/motion.1.log)
[boot(1).log](https://github.com/ccrisan/motioneyeos/files/6039628/boot.1.log)

starbasessd, 2021-02-25:
What are the network settings on the PC you're using, what Subnet mask,
what IP subnet?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zacharym8, 2021-02-25:
Main computer is using windows and is connected by ethernet to the router (ipconfig provides the following):
   Connection-specific DNS Suffix  . : localdomain
   IPv4 Address. . . . . . . . . . . : 10.0.0.100
   Subnet Mask . . . . . . . . . . . : 255.0.0.0
   Default Gateway . . . . . . . . . : 10.0.0.1

The **pi with motioneyeos** that is experiencing this issue shows the following on an "ip a":
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether b8:27:eb:e8:00:f6 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.251/8 brd 10.255.255.255 scope global wlan0
       valid_lft forever preferred_lft forever

And the route command shows:
default via 10.0.0.1 dev wlan0

The **pi that is raspbian with motioneye loaded** on it shows the following with the same two commands:
2: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:6d:a2:f8 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.216/8 brd 10.255.255.255 scope global dynamic noprefixroute wlan0
       valid_lft 6791sec preferred_lft 5891sec

and 
default via 10.0.0.1 dev wlan0 proto dhcp src 10.0.0.216 metric 302

starbasessd, 2021-02-25:
If you install mEOS on the Pi in question, with wpa_supplicant.conf and
ssh.txt, then boot it up, connect to it, Do NOT make any changes, tell it
to reboot, does it come back up normally?
If it does, ssh or console, and run commands:
vcgencmd get_throttled
and
vcgencmd measure_temp
and report results, please

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

zacharym8, 2021-02-25:
So I wiped the card, reloaded 20190911 onto it, put wpa_supplicant and ssh in the boot, and booted it up.
It came up fine but same issue consistently on the first boot as well, I could not ping or ssh the device, but once I used the ping command on the pi to my main computer, my main computer could then ping and ssh in. I have made no changes, just pinged it and connected to it for 5 seconds through ssh.
Then I issued the reboot now command, it rebooted perfectly fine, and had the same issue. I did not do the temp fix where I ping outward so that I could do the commmands you posted while it was still actively in the problem state. The results are as follows:

**vcgencmd get_throttled:** 
throttled=0x0

**vcgencmd measure_temp**
temp=41.2 C

starbasessd, 2021-02-25:
Is there a reason why you aren't using a newer version?
I'm sitting here with 3 PiZeroWs, one each of 20200203, 202020606, and
dev20201026.
All are set to DHCP, with the same wpa_supplicant.conf file and ssh.txt
file.
My wpa_supplicant.conf file contains:
---
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="my_ssid"
    psk="my_password"
    key_mgmt=WPA-PSK
}
---
All 3 connected to my Access Point/Router (set to pass-through to let my
primary router do DHCP)
I then connected to all 3 and did Reboot from the WebGUI
All 3 reconnected, no issue
I connected via SSH, and rebooted, again, no issue.
I can ping from my router or PC or VM Server, all no issues.
My DHCP service issues 192.168.X.Y /24 with DNS 192.168.X.1, and 4.2.2.4,
and Primary Gateway 192.168.X.1

This works for All my Pis, whether running motionEyeOS, RaspberryPiOS,
Ubuntu 20.04 Desktop, my own dev OS, whatever.
Also works for all my VBox VMs, my Docker VMs, my KVM VMs, and my Win10 VMs.
I am not sure what is different on your network, but if the PI is up, with
a valid IP & subnet mask, and default Gateway (matching those of the PC)
Unless there is a 'smart' firewall on the wifi AP or router, I can't think
of anything else.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zacharym8, 2021-02-25:
I was using the 20190911 version because earlier when you requested I try it I just reloaded it to hopefully reduce down on variables involved. I just retried it with the 20200606, used the exact wpa_supplicant you posted above, and its still the same story, cannot access the device until I ping my computer from the pi itself. I also loaded on another zero w the Raspbian OS and then loaded motioneye on to that one also and it still works right off the bat that way. Is there anything that motioneyeos sends out traffic wise that is different from a raspbian+motioneye build that a router could be stopping? 

Though the part that is confusing for me for if it is an issue with my router is why the raspbian+motioneye ones work, but the motioneyeos's don't, and how if I ping a computer from the pi then it immediately is reachable and back to normal.

My exact network setup is a pass-through access point connected back to a switch, which is where my main router, computer, and other hardwire devices are plugged in. I have many servers and VMs running all along the network and they all function fine.

starbasessd, 2021-02-25:
Unfortunately, no, I don't have a clue as to why you're having issues.
Usually it's related to a setting in DHCP (maybe a comma instead of a
period kind of thing).
Here's a thought:
Set a static IP in the /boot partition prior to first boot.
Do you have an address (not in your DHCP range) available in the same /24
address space?
If so, I can give you the files needed.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zacharym8, 2021-02-26:
I tried to look around and find a guide, but couldn't seem to find any, so if you have files I could use for trying that I will get that done as soon as I can. I do have a range of addresses outside of DHCP for specific things so I can try that as soon as I figure out how to get a static IP set through the /boot partition on initial set up.

starbasessd, 2021-02-26:
static_ip.conf (in the zip attachment) sets the static IP. Make changes as needed, then include in the /boot partition prior to first boot.
Instructions are on the right side of the main Wiki of motionEyeOS under Configuration:
https://github.com/ccrisan/thingos/wiki/static_ip.conf

[static_ip.zip](https://github.com/ccrisan/motioneyeos/files/6052952/static_ip.zip)

zacharym8, 2021-02-26:
I went ahead and reflashed 20200606, added in wpa_supplicant, ssh, and the static_ip files to boot, and booted it up. All the configurations took hold, it can ping google and such, but I am still unable to ssh/ping/connect to it, but the moment I ping my computer from it its back to working.

starbasessd, 2021-02-26:
I really don't know, then. I'll leave it here, obviously, hopefully someone
else has a valid idea...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?