motioneye-project/motioneyeos #2681: Auto-reset system configuration

sagitt, 2021-02-25:
### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Rpi Zero W

### Camera

I am using the following type of camera: Fast camera with CSI camera

My camera model is: https://it.aliexpress.com/item/33003508360.html?srcSns=sns_Copy&spreadType=socialShare&bizType=ProductDetail&social_params=60033698801&tt=MG&aff_fsk=_vX8QyD&aff_platform=default&sk=_vX8QyD&aff_trace_key=86db5e32c661477f8dcafa7015bb7d51-1614238809199-05065-_vX8QyD&shareId=60033698801&businessType=ProductDetail&platform=AE&terminal_id=487ddfbffdc44b2aa236f2ec45c96948

### Network Connection

My motionEyeOS unit is connected to the network via: Wi-Fi

### Problem

I have an RPi Zero W configured as Fast Camera on front door and all works fine, but sometimes i see the camera not available in the network... nothing change removing power and restoring it. This happened like 30 times in a year.

Everytime happen this, if i re-flash motioneyeos (with writeimage.sh) on the SD, all starts again ad works.

So i tried to replace RPi Zero W (maybe hardware fault?)...
So i tried to replace SD card with different types (Samsung, SanDisk Extreme, Extreme Pro, Max Endurance... different sizes - now using Max Endurance).. same result.

I done some tests and testing the SD on another RPi Zero W with a monitor attached, i see that the hostname changes to the default and it can't connect to the network.. like auto-resetted itself.

What can do this? i'm now trying to change network link watch from 60s to 3600s... is possible is it?

The device is under APC ups so no power fault. Temperature range: 40-60°

Please help me. Thanks

starbasessd, 2021-02-25:
I would check the messages log for the temps it reports. Getting at or
above 60`C indicates it may be throttling, which causes problems with
wifi/network. You can run the command
vcgencmd get_throttled
and if the response is anything but 0x0 it's having issues with low voltage
or temperature.
UPS doesn't guarantee correct voltage from the power supply for the Pi.
---
If could, ssh or console in, and run the command
logrotate /etc/logrotate.conf
then wait 24-48 hours, and post all the logs here.
---
Why would you need to re-image after the failure? Would it not boot anymore?
---

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sagitt, 2021-02-25:
> I would check the messages log for the temps it reports. Getting at or
> above 60`C indicates it may be throttling, which causes problems with
> wifi/network. You can run the command
> vcgencmd get_throttled
> and if the response is anything but 0x0 it's having issues with low voltage
> or temperature.
> UPS doesn't guarantee correct voltage from the power supply for the Pi.
> ---
> If could, ssh or console in, and run the command
> logrotate /etc/logrotate.conf
> then wait 24-48 hours, and post all the logs here.
> ---
> Why would you need to re-image after the failure? Would it not boot anymore?
> ---
> 
> 
> <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
> Virus-free.
> www.avg.com
> <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
> <#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>
> 
> 
> > motionEyeOS Version
> >
> > I am running motionEyeOS version: 20200606
> > Board Model
> >
> > I am using the following board/model: Rpi Zero W
> > Camera
> >
> > I am using the following type of camera: Fast camera with CSI camera
> >
> > My camera model is:
> > https://it.aliexpress.com/item/33003508360.html?srcSns=sns_Copy&spreadType=socialShare&bizType=ProductDetail&social_params=60033698801&tt=MG&aff_fsk=_vX8QyD&aff_platform=default&sk=_vX8QyD&aff_trace_key=86db5e32c661477f8dcafa7015bb7d51-1614238809199-05065-_vX8QyD&shareId=60033698801&businessType=ProductDetail&platform=AE&terminal_id=487ddfbffdc44b2aa236f2ec45c96948
> > Network Connection
> >
> > My motionEyeOS unit is connected to the network via: Wi-Fi
> > Problem
> >
> > I have an RPi Zero W configured as Fast Camera on front door and all works
> > fine, but sometimes i see the camera not available in the network...
> > nothing change removing power and restoring it. This happened like 30 times
> > in a year.
> >
> > Everytime happen this, if i re-flash motioneyeos (with writeimage.sh) on
> > the SD, all starts again ad works.
> >
> > So i tried to replace RPi Zero W (maybe hardware fault?)...
> > So i tried to replace SD card with different types (Samsung, SanDisk
> > Extreme, Extreme Pro, Max Endurance... different sizes - now using Max
> > Endurance).. same result.
> >
> > I done some tests and testing the SD on another RPi Zero W with a monitor
> > attached, i see that the hostname changes to the default and it can't
> > connect to the network.. like auto-resetted itself.
> >
> > What can do this? i'm now trying to change network link watch from 60s to
> > 3600s... is possible is it?
> >
> > The device is under APC ups so no power fault. Temperature range: 40-60°
> >
> > Please help me. Thanks
> >
> >
> 
> 
> Thanks
> 
> Kevin Shumaker
> 
> 
> Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>
> 
> N38° 19' 56.52"
> W85° 45' 8.56"
> 
> Semper Gumby
> “Don't tell people how to do things. Tell them what to do and let them
> surprise you with their results.” - G.S. Patton, Gen. USA
> Ethics are what we do when no one else is looking.
> Quis custodiet ipsos custodes?
> “There is no end to the good you can do if you don’t care who
> gets the credit.” - C Powell
> You know we're sitting on four million pounds of fuel, one nuclear weapon
> and a thing that has 270,000 moving parts built by the lowest bidder. Makes
> you feel good, doesn't it?

Result is 0x0. I'm using professional 5v power supply on DIN RAIL.

I done log command. I'll wait for it.

Where i found the final log file?

I have to re-write image because it boot ok, but no confg. See picture below. Hostname meye-xxxx. If i boot with configured sd i see hostname that i setted in the web ui

[image]

starbasessd, 2021-02-25:
Log files can be found in the GUI at Settings, Expert Settings
Run the vcgencmd commands again after you've been up 24-48 hours, too, and
post here.

What's up with the screen shot? It was taken after a failed upgrade from an
ancient (relatively speaking) upgrade.
If that is current, then your SDCard is SERIOUSLY corrupted or broken. You
need to install either of these from scratch:
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz
or
https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi-20200606.img.xz
for the PiZero. Don't use the writeimage.sh, use either the RPiImager from
the RaspberryPiFoundation (available on raspberryPi.com downloads) or
Win32DiskImager available here:
https://sourceforge.net/projects/win32diskimager/

Result is 0x0. I'm using professional 5v power supply on DIN RAIL.

> I done log command. I'll wait for it.
>
> Where i found the final log file?
>
> I have to re-write image because it boot ok, but no confg. See picture
> below. Hostname meye-xxxx. If i boot with configured [...]

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

starbasessd, 2021-02-25:
BTW, just because the power source is a "professional 5v power supply on DIN RAIL." doesn't mean it isn't off just a bit, or that the USB cable isn't poor quality, or if you are feeding into the GPIO pins on the Pi, you are bypassing the built in voltage regulator, and could be having other issues.

sagitt, 2021-02-25:
> BTW, just because the power source is a "professional 5v power supply on DIN RAIL." doesn't mean it isn't off just a bit, or that the USB cable isn't poor quality, or if you are feeding into the GPIO pins on the Pi, you are bypassing the built in voltage regulator, and could be having other issues.

Yes, I'm using the gpio for powering it! But i'm using good quality cable. If the problem is this, maybe i can change with usb instead gpio.. but before do it, i hop to undestand the real cause of the problem!

Where i can read the generated log?

Thanks for all!

starbasessd, 2021-02-25:
Log files can be found in the GUI at Settings, Expert Settings
They are also in /var/log ( grab the ones without the number at the end)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sagitt, 2021-02-27:
> Log files can be found in the GUI at Settings, Expert Settings They are also in /var/log ( grab the ones without the number at the end)
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

i'm not able to access to log file... after 1 days same results.

this is after failure:

[image]

this normal boot:

[image]

i don't understand what can do this.

starbasessd, 2021-02-27:
You need to re-image. The system is/has crashed after an attempted upgrade
from a copy that is too old to upgrade.
I suggest a new download from the Releases. Do not use writeimage.sh, get
Raspberry Pi Imager, Win32DiskImager or Balena Imager.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sagitt, 2021-02-27:
> You need to re-image. The system is/has crashed after an attempted upgrade from a copy that is too old to upgrade. I suggest a new download from the Releases. Do not use writeimage.sh, get Raspberry Pi Imager, Win32DiskImager or Balena Imager.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

i downloaded it from releases
used: 20200606
motioneyeos-raspberrypi-20200606.img.xz

how is possible this?

starbasessd, 2021-02-27:
If the image is not being written correctly, it's not overwriting the
system correctly. I have a USB memory stick that acts like that.
Delete all the data and partitions, and start over, or use a different
SDCard.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sagitt, 2021-02-27:
> If the image is not being written correctly, it's not overwriting the system correctly. I have a USB memory stick that acts like that. Delete all the data and partitions, and start over, or use a different SDCard.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

i'll try again with balena, but every time i flash the system i totally reset the SD card, and done the same with differente cards (new cards too).

Thanks

sagitt, 2021-10-05:
after months i tried to replace rpi zero w, sd card (2 different max endurance card), usb switched from gpio to usb... same result... sometime system reset and the only solution is re-flash card.

starbasessd, 2021-10-05:
Card is corrupted. Usually caused by overheating or low voltage, can be caused by disk full (but not often). Do you have a Linux box to read the card for the logs?

sagitt, 2021-10-05:
yes i can read it on linux virtual machine for example. anyway no lower voltage, i checked it and is stable 

max temperature 50° (i checked it for days)
throttling ever: throttled=0x0

the next time i can check in logs, where i can found something useful for this problem? what file?

thanks

starbasessd, 2021-10-05:
The 'big 5' logs: dmesg, messages.log, motion.log, motioneye.log, and boot.log, all under /data (3rd partition) /data/var/log
Also check free space on all 3 partitions:
/boot, /root, and /data
df -h
the partitions are usually mmcblk0p1, mmcblk0p2, and mmcblk0p3 (unless the machine reading them used an SDCard to boot from)