motioneye-project/motioneyeos #2683: unsuported cam

Steven0972, 2021-02-27:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
yes
I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
yes
### motionEyeOS Version

I am running motionEyeOS version:

motionEye Version | 0.39.3
-- | --
Motion Version | 4.1.1
OS Version | motionEyeOS 20190119

.

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: Ip camera

My camera model is:old ip cam sitcom LN-400 an IPC1002.

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
nothing relevant

### Log Files

I consider the following log files relevant to this issue:

when I tried to insert then to motion I get unsupported camera ( network and mpeg cam)
is there any way to get these cam's working in motion eye

starbasessd, 2021-02-27:
Please try a newer version of motionEyeOS, 20200203, 20200606, or
dev20201026. The version you are attempting to use is 2+ years old, and
there have been MANY improvements since then, including using newwer
kernels with more cameras supported.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Steven0972, 2021-02-27:
Just upgrade to latest version
Still the same , any suggestions 

Regards Steven

starbasessd, 2021-02-27:
Please confirm the camera makers and models, I cannot locate any info on "
sitcom LN-400 an IPC1002"

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Steven0972, 2021-02-27:
This one

And this one

starbasessd, 2021-02-27:
Nothing came across.

Steven0972, 2021-02-27:
so there is no way to get them working in motion?
other programs recognize the camera en you can also watch them in a brouwser. but not in motioneye.
thats a pitty.
i like motion and i want to keep using it, but if i dont get the cams workin i'll have to look else where. 
regards

Steven Lentzen
steven.lentzen@telenet.be

starbasessd, 2021-02-27:
Maybe. I asked for better information on the maker and models of the
cameras, I couldn't find any info.
Does it work with Raspberry Pi OS?
What URL do you use to view with a browser?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Steven0972, 2021-02-27:
didn't you get the screenshots?
it is a vilar ipc1002 ipcam and a sitecom LN-400 wired ipcam. that is the infomation i've got from the camera's.
url is the ip adres of the camera
and it works with pi os, the url even works in openhabian, ubuntu, windows, mac, ios and android
works in safari firefox chrome,...
but not in motioneye. " unsupported camera" it says

MVG
Steven Lentzen
steven.lentzen@telenet.be

starbasessd, 2021-02-27:
No, they didn't come across.
For the Vilar URL are you using
http://[ip_address]/cgi.bin/sf.cgi
and for the sitecom URL are you using
http://[ip_address]/video.cgi?resolution=VGA

starbasessd, 2022-01-19:
https://github.com/ccrisan/motioneyeos/issues/1778

starbasessd, 2022-01-19:
No response in more than 60 days. If you are still having issues, please open a new issue.

closed 2022-01-19 by starbasessd