motioneye-project/motioneyeos #2684: Issue with setting /data/log to /tmp or to a usb device

queenar, 2021-02-28:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-raspberrypi2-20200606

### Board Model

I am using the following board/model: RPI2

### Camera

Standard rpi-noir cam v2, but got no problem with camera anyhow :).

### Network Connection

Wifi via wpa_supplicant.conf, works fine.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

* USB WiFi adapter - works fine
* usb device from sandisk. It is recognized correctly.

### Log Files

None for now as they are not written (read-only), and only printed on monitor while booting.
 

Hi everyone! And first of all thanks for this great piece of software :-) !

For getting a better sd card timespan I started sometime ago to only boot from sd card and put all other vital data to an usb device for the most of my raspberries.
I did not do that at first with motioneyeos, but however one sd card got "eaten" up last week and was not recoverable (fsck failed). But that is fine it now runs on another sd card.

Now for my problem/question. I like to store the logs (and images) on an external usb device. Which works fine for the images, but not the log files.

logs to ram:
I followed the wiki article on how to reside the logs in RAM. But for me this did not work, upon rebooting the device it won't come back on, but states that the `/var/log` is read-only. So for me the part in the wiki does not work:
https://github.com/ccrisan/thingos/wiki/Logs

data on usb device:
This would be my preferred way of doing it. I tried to mount the usb device as `/data` in the fstab.disk:
`/dev/sda1       /data                 ext4     defaults,noatime  0      0`
This worked like a charm! But only for the first boot :D.
After rebooting once, more errors appear on how `/data` is read-only which makes the device unusable.
`Mounting filesystems: mount: mounting /dev/sda1 on /data failed: No such file or directory`
And then some other errors about not able to mount some mountpoints, which is fine as he does not even have /data.
I do not quite get what happens here, why it works on the first boot but not after reboot. The usb device is set up correctly, all folders seem to be there, but after the reboot, read-only again?

Can someone help me out here? I also tried to symlink `/data/log` to `/media/sda1/log` which would be fine for me too, but again just read-only errors.
What am I missing here?

starbasessd, 2021-02-28:
Logs are available in WebGUI. Settings, Expert.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

queenar, 2021-02-28:
Hi and thanks for your comment. I know that I can find the logs on the web ui, but I have got no web ui as the /data partition does not get mounted and there is no way to even access the ui :).
But for the described problem I doubt that logs are even necessary or explaining anything.

Anyhow my topic is more of a question, but a problem as I changed the fstab.disk here. Without changing anything, the system runs smoothly.

starbasessd, 2021-02-28:
If using motionEyeOS, it has 3 partitions and mounts
/dev/mmcblk0p1 as /boot
/dev/mmcblk0p2 as / (root)
/dev/mmcblk0p3 as /data and /home/ftp/sdcard and /home/ftp/storage
It mounts the overlay file system for  /usr /var/log and /var/lib
the /dev/mmcblk0p3 and overlay filesystems are mounted rw, and you can
write data as needed, and the logs are written and saved.
You should see the logs in /data/log folder, and they can be VERY helpful
for a lot of issues.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

queenar, 2021-02-28:
Hi starbasessd, thanks again for your answer and I can only confirm what you say, logfiles are crucial - but again, I do not have any when I mount /dev/sda1 as /data as it is mounted read-only by default and no logs can be written anywhere, like stated in my first post. I do not know what else to say ... I know about the partitioning layout and yes my sdcard has these 3 partitions - but I try to use /dev/sda1 instead of xxx0p3 for /data.
Again, I am thankful for this software and I do not want to upset any one with a issue not and providing the vital infos. I hope my first post describes what I am trying to achieve. English is not my native language, so if anything is unclear, please let me know.

starbasessd, 2021-02-28:
What changes are you making to mount /dev/sda1 as /data ?
I believe, due to the overlay file system, you need to remount all the links to the /data on mmcblk0p3 to your sda1, as well as the overlays or you will break things.
You would need to mount /dev/sda1 in fstab files in /etc (there are 4) , in the same manner as the mmcblk0p3 and overlays
You would need to enable rw for /dev/mmcblk0p2 (/) then edit the 4 fstab files.
Unless you build your own version using the build files, all this would have to be done after first boot.
I can point you in the right direction, or you could make this a 'feature request' (no telling when it would be added) or you could do a pull request to modify the setup files...

cap9qd, 2021-02-28:
So what I am reading is that you want to mount your USB drive as /data instead of the 3rd partition on the SD card?  I tired to do something similar with my Odroid hardware and was unsuccessful.

I tried this with my Odroid HC1 that has a built in USB3-SATA hardware.  This worked in Ubuntu but I was unable to get it to work in MotionEyeOS. There are guilds to do this in Ubuntu 18.04 and 20.04 for the Odroid HC1/2 but I was unable to get the same steps adapted to MotionEyeOS after first boot.  

The issue I had and was unable to resolve is that the /dev/sda1 device is not detected early enough in the boot sequence when fstab.disk /data is being mounted; my /dev/sda1 was only detected after udev started in the boot process. I think this is why you are seeing the "file not found" error when trying to mount /dev/sda1. 

I built my "own" image from source but still was unable to get this part to work but was able to add python and other programs already setup in BuildRoot. 

You can also change the boot arguments to remove the "quiet" and "loglevel=1" arguments for the kernel start commands to get the boot process to print much more information to the screen since you don't have logs during this time while working to get /data/ mounted; I think it is the same for the RPi hardware in the cmdline.txt file on the boot partition.  Since the boot partition is FAT32 or FAT16 then you should be able to make this change with a Unix file endings aware text editor on a windows machine by mounting the SD card in Windows.

I guess this isn't a lot of help but thought I would share my experience.  I would be very interested if you were able to solve this problem so please keep us posted!

queenar, 2021-03-01:
Hi you two!

Yes  cap9qd, that's what I wanted to do and I think your points are indeed helpful, I did not think about setting the loglevel a bit higher, will do this when I test again! :)
The thing what bothered me was that `/dev/sda1` was sucessfully mounted as /data on the very first boot, when the system is set up the first time. But when this is done (`etc` files written, and so on), the next reboot ends in the read-only problem.
So I guess there are more places where this has to be changed - or even built with the own build environment, like you mentioned. But I hope that's not needed after all.

And now the questions from starbasessd! What did I do.. I will try to recap and write it down here :).
I use a Linux machine for writing the images, so first of all: use `dd` to write the image to my sd card.
Then I mount the first partition boot, and add my `wpa_supplicant.conf` file to get the network to run directly.
Next step, mount the second partition, cannot recall the name, but the one with the real system on it (/dev/mmcblk0p2), which gets mounted `ro` when the system runs. And as I understood it, some files from there are copied to the third partition, which is data on the first boot (, or when /`data` is detected empty I guess, to reset everything). The `etc` dir with some files and some other folders.

So my idea was to edit the `etc/fstab.disk` file before the first boot, as this seemed to be the right place where the third partition gets bound da `/data`.
So this is the part where I replaced the mmcblk0p3 partition with `/dev/sda1`.

But like cap9qd pointed out, this could as well be something with udev. On the other hand, isn't udev something that runs really early in the boot process?
The overlays could be a problem, but I think as they point to `/data` it should be just fine when just the entry `/dev/mmcblk0p3  /data ` is changed, what do you think? (The output is from a running instance only on sdcard)
```
[root@meye ~]# cat /etc/fstab.*
# <file system> <mount pt>            <type>   <options>         <dump> <pass>
/dev/mmcblk0p1  /boot                 vfat     ro,defaults       0      0
/dev/mmcblk0p3  /data                 ext4     defaults,noatime  0      0
# <file system> <mount pt>            <type>   <options>         <dump> <pass>
/data/output    /home/ftp/sdcard      rbind    rbind             0      0
/data/media     /home/ftp/storage     rbind    rbind             0      0
# <file system> <mount pt>            <type>   <options>                                                                <dump> <pass>
overlay         /usr                  overlay  lowerdir=/usr,upperdir=/data/usr,workdir=/data/.overlay-usr              0 0
overlay         /var/log              overlay  lowerdir=/var/log,upperdir=/data/log,workdir=/data/.overlay-log          0 0
overlay         /var/lib              overlay  lowerdir=/var/lib,upperdir=/data/varlib,workdir=/data/.overlay-varlib    0 0
# <file system> <mount pt>            <type>   <options>         <dump> <pass>
proc            /proc                 proc     defaults          0      0
devpts          /dev/pts              devpts   gid=5,mode=620    0      0
tmpfs           /dev/shm              tmpfs    mode=0777         0      0
tmpfs           /tmp                  tmpfs    mode=1777         0      0
sysfs           /sys                  sysfs    defaults          0      0
configfs        /sys/kernel/config    configfs defaults          0      0
```

I will try it with a more verbose boot and come back in touch with you. It may well be that I can do the testing not until the weekend. Thanks for your input!

starbasessd, 2021-03-01:
A somewhat helpful link:
https://github.com/ghollingworth/overlayfs
RaspberryPiOS has it as an option to play with it, too.
Depending on which FS is the 'under' and which is the 'over' you can set it
up as they have in motionEyeOS where you're not totally hitting the / root
system (the ro limited) and still mod the /data (over) where the files are
rw. That way, all the mods are in the /data/etc, /data/var, /data/usr, but
are transparent to the OS. No need for lots of simlinks and re-directs, and
the 2 system partitions are safely ro.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

queenar, 2021-03-08:
Okay, sadly just a short update from my side. It still does not work.

Yesterday I had two hours to play around with udev, fstab and overlays a bit. But nothing I did really helped at all. Every time I get back to the point where something (or some mounting at least) fails and he only mounts /dev/root as / in ro (which is fine) but without any other writeable drive on /data.
Which is a bit frustating and time-consuming, as I have to rewrite the sdcard everytime this happens. When I only wipe the 3rd partition on sdcard (and revert my changes from any files on the other partitions of course) I hoped that the system would fill itself again (recovery). But for my observation it does for some part, but without any chance I could log in via terminal nor ssh (wrong user/password).
I had to figure out udev a bit. Had many failures here, but in the end just adapted the existing usb-storage rules to symlink my /dev/sda1 to /dev/usbdata and mount it to /data1. Worked fine.
But when trying to mount /dev/sda1 directly to /data with udev, it broke again.

As I tried out many different things, I fear I cannot really give you a whole overview with commands, changes and files I modified.

The last thing I tried out was the overlay part. I thought it could be worth trying out mounting sda1 as /data1 and then use the overlay fs to go on top of /data:
```
[root@meye-c2ccf6bf ~]# mount; ll /dev/usbd*
/dev/root on / type ext4 (ro,relatime,norecovery)
devtmpfs on /dev type devtmpfs (rw,relatime,size=378192k,nr_inodes=94548,mode=755)
proc on /proc type proc (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /dev/shm type tmpfs (rw,relatime,mode=777)
tmpfs on /tmp type tmpfs (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p1 on /boot type vfat (ro,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p3 on /data type ext4 (rw,noatime)
overlay on /usr type overlay (rw,relatime,lowerdir=/usr,upperdir=/data/usr,workdir=/data/.overlay-usr)
overlay on /var/log type overlay (rw,relatime,lowerdir=/var/log,upperdir=/data/log,workdir=/data/.overlay-log)
overlay on /var/lib type overlay (rw,relatime,lowerdir=/var/lib,upperdir=/data/varlib,workdir=/data/.overlay-varlib)
/dev/sda1 on /data/media/sda1 type ext4 (rw,relatime)
lrwxrwxrwx    1 root     root           4 Jan  1  1970 /dev/usbdata -> sda1
```
```
[root@meye-c2ccf6bf ~]# cat /etc/udev/rules.d/10-data.rules 
KERNEL=="sda1", SUBSYSTEMS=="scsi", SYMLINK="usbdata"
```
```
[root@meye-c2ccf6bf ~]# cat /etc/fstab.disk 
# <file system> <mount pt>            <type>   <options>         <dump> <pass>
/dev/mmcblk0p1  /boot                 vfat     ro,defaults       0      0
/dev/usbdata    /data1                ext4     defaults,noatime  0      0
/dev/mmcblk0p3  /data                 ext4     defaults,noatime  0      0
```
```
[root@meye-c2ccf6bf ~]# cat /etc/fstab.overlay 
# <file system> <mount pt>            <type>   <options>                                                                <dump> <pass>
overlay         /data                 overlay  lowerdir=/data,upperdir=/data1,workdir=/data1/.overlay-data              0 0
overlay         /usr                  overlay  lowerdir=/usr,upperdir=/data/usr,workdir=/data/.overlay-usr              0 0
overlay         /var/log              overlay  lowerdir=/var/log,upperdir=/data/log,workdir=/data/.overlay-log          0 0
overlay         /var/lib              overlay  lowerdir=/var/lib,upperdir=/data/varlib,workdir=/data/.overlay-varlib    0 0
```
That does not work at all. /data1 is there, but the /data overlay is not mounted correctly. Error: `Can't create directory /data1/.overlay-data: Read-only file system`.
And that is the point I do not really understand, as I can just fine touch /data1/test :-). But for overlays and udev, I am not really into that, just starting. So maybe I got something wrong here. Any suggestions maybe?

So this is all for now, I think I will start digging again in a few days, when I find the time.

starbasessd, 2021-03-08:
The only folder tree you should be trying to modify is /data
the /data/etc/fstab.xxx files you need to modify are in there.
Request: downgrade to 20190911 or upgrade to dev20201026

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

queenar, 2021-03-08:
The folder /data/etc has no fstab.* files in it. The ones used while the system boots reside on the ro second partition of the sdcard which is mounted from /dev/root to /. So I think it is just fine to edit these (either after dd of the image, or with mount -o rw,remount /dev/root on the system running).
Or when I create /data/etc/fstab.disk, will this be taken instead of /etc/fstab.disk? Is that what you meant?

Why the request to downgrade or upgrade to a dev version? This is not a bug after all and I guess the system design / os design did not really change this much? But I will have a look at the latest dev version then.

starbasessd, 2021-03-08:
Hmmm...
Cannot seem to get an overlay for an overlay file system
changed / to rw
I created /sda1, same parameters as /data
I mounted /dev/sda1 to it in fstab.disk as rw
rebooted, works
Tried to get /data as lower, /sda1/data as upper, but keep getting error
Mounting overlay filesystems: mkdir: can't create directory 'O' : Read-only
file system
researching.
next test will be to change existing fstab.overlay to /sda1

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-08:
No joy. Simply modding a few config files isn't working.
There may be a number of hard coded links within the system to prevent it
from working, or being easily changed.
I will leave this here for others to look at.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

queenar, 2021-03-09:
Thanks for also testing it out! A bit glad I did not make a some mistakes and you had a same experience as I had with editing the files. I also thought there could be some paths hardcoded elsewhere.
Maybe someone can enlighten us here? :smile:

starbasessd, 2022-01-19:
No response in more than 60 days. If you are still having issues, please open a new issue.

closed 2022-01-19 by starbasessd

bezik46, 2025-05-18:
Would it really be too difficult to have proper instructions for something "as simple" as being able to edit config files?