motioneye-project/motioneyeos #2685: Take pictures

dzikija, 2021-02-28:
Hi, is it possible to add the option to take pictures in full camera matrix resolution and full quality? Thanks for the information.

starbasessd, 2021-03-01:
It takes pictures at whatever resolution & quality you have it set to. You can set Image Quality to 100% and not get any jpg compression.

dzikija, 2021-03-02:
Thank you for the information, after work, I will check everything again. But it seems to me that only resolution max up to 1920 × 1080 (the same like Full HD Video). By using: raspistill -o test.jpg the photo is 5mp (2592x1944px). In the future, I would like to use an 8MP camera and  take pictures of this resolution.

dzikija, 2021-03-02:
Someone has already asked about it:
https://github.com/ccrisan/motioneyeos/issues/513
I meant that too.