motioneye-project/motioneyeos #2686: No media files info after months of correct saving of mp4 and jpg

wkulesza, 2021-03-01:
Hi,
First of all congrats on great tool.
We used it to monitor our office when we work mostly from home now.
It's working fine and I can see live view and also get to see saved motion triggered mp4 and jpg.

Suddenly today I got Camera1 (no media files) on both video and photo section.

Settings show there's 80% of sd card filled at the moment so space is still available.

Any idea what would cause this? I need to access footage but not at the office at the moment.

starbasessd, 2021-03-01:
Can you ssh into it? If yes, do a df -h and check space available in each of the 3 partitions:
/
/boot
/data
Sounds like something got corrupted or changed.
Can you force it to save a snapshot?
I hate to recommend rebooting if you can't get to it to reset or re-image if needed.

starbasessd, 2021-03-01:
BTW, you didn't supply any answers to the standard questions:
Which version of motionEyeOS?
Which Pi or similar board?
For me:
How big an SDCard are you using?
When was the last time it was rebooted?

wkulesza, 2021-03-02:
Hi,
yes i can login and df -h shows:
[image]

Now in office i restarted the pi - I can access the images and videos; but between yesterday around 8:57am and today (when i restarted) there are no images available.

motionEye Version | 0.42.1
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606

Im using Rpi3B.
The SD card is 8gb
It was never rebooted.

What are the best practices for maintenance - one should schedule a reboot every now and then and manually deleted images from the card, or will they be deleted (from the oldest) automatically when sd card is full ?

starbasessd, 2021-03-02:
I cannot give a 'best practice number, because every situation is different. 
The card is what I would consider small, and would be checking it more often, than say, a 32GB or 64GB SD Card, or USB Thumb drive for movies and snap shots, especially if it gets below 1GB available.
You can set it to automatically delete video and snap shots daily, weekly, monthly or yearly, but not by how full the drive is, unfortunately. There are a few hacks floating around to try to keep it clear, but nothing 'official' It's in the 'most wanted' enhancements list.
It is easy between logs and videos and snap shots for something to 'run away, and rapidly fill a small drive. 
As to rebooting, it's easy to add a crontab job to force a reboot every once in a while (weekly, monthly, whatever is comfortable for you) automatically, or as a sysAdmin chore, manually however often you decide is appropriate. 
On my production Hub, I reboot weekly. I use a NAS for primary and secondary storage, have a USB drive for local stuff if needed. I clean my drives weekly of old files manually weekly as part of my sysAdmin chores checklist. I have several 'stand by' systems (VMs, and PiHubs) for testing, doing these support issues, etc. that I can activate if the production server were to go down. If you need more info, please ask.

wkulesza, 2021-04-12:
@starbasessd
thanks for pointers.
I have created the following crontab but noticed today that uptime command shows 13 days uptime, while it should reboot over sunday ?
```
[root@meye-53924c39 ~]# crontab -l
30 1  *    *    7 /sbin/shutdown -r now
```
does crontab work out of the box in motioneyeos or needs to be modified/enabled ?

starbasessd, 2021-04-12:
/sbin/shutdown doesn't exist in motionEyeOS. use /sbin/reboot

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2022-01-19:
No response in more than 60 days. If you are still having issues, please open a new issue.

closed 2022-01-19 by starbasessd