motioneye-project/motioneyeos #2687: Camera feed freezes after a few seconds on varying resolutions/FPS

tdickie77, 2021-03-01:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running this as a Docker container.  Not sure the absolute version, but I get this from the UI

motionEye Version | 0.42.1
Motion Version | 4.2.2
OS Version | Debian 9.12

Main OS info:
Operating System: Raspbian GNU/Linux 10 (buster)
Kernel: Linux 5.10.11-v7l+
Architecture: arm

### Board Model

I am using the following board/model: Compute Module 4.

### Camera

I am using the following type of camera: 
V4L2 using a Mokose UC40A camera (USB)

### Network Connection

My motionEyeOS unit is connected to the network via ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
N/A

### Log Files

I consider the following log files relevant to this issue:

Unable to locate logs.  No Expert mode seen and running as a container. 

Issue:

I currently have the container set to autostart on port 8765 per a document I found here.  The camera feed will show and then will stop.  The text overlay shows the time just stopping.  I have tried various settings, such as resolution from 1920x1080 to 640x480, motion capture on/off, FPS from 1 to 30.  Unable to figure out why it just stops.  May have to do with the camera itself or how the camera is being used through the OS.  Some help in this regard would be great.  I looked through the forum, but am unable to find anyone else with this issue from what I can tell.  Thanks in advance.

starbasessd, 2021-03-01:
Better to open the issue in the motionEye github here:
https://github.com/ccrisan/motioneye/issues
I don't have a compute module to test with.
If you close this one and open in the motionEye github, I'll be happy to try to help.

tdickie77, 2021-03-01:
Hey @starbasessd I appreciate the direction.  Actually, I just found out that another application i am running (just installed) is causing a conflict and had to uninstall Motioneye.  This was not the cause of the issue, but I won;t be able to troubleshoot further due to the amount of dependence my system has on the new app.  However, I do really love this MotionEye application and plan to use it for other scenarios.  Will just have to put it on the backburner for now.  I really appreciate your quick response and hope I can set this back up in the very near future for testing again.

closed 2021-03-01 by tdickie77