motioneye-project/motioneyeos #2688: IR LEDs (Lisiparoi)

mikefloyd91, 2021-03-02:
Hi All,

Relatively new to all of this and keen to learn but having some issues that I cannot find elsewhere in the project. Hoping to set up a RP with motioneyeos as a Birdbox camera.

I have a Raspberry Pi NOIR camera successfully connected up and have wired up a Lisiparoi IR LED unit to the GPIO Pins. However, the LEDs don't turn on at boot. Do I need to add a line of code to the boot file, or can it be controlled via the web interface?

Apologies is this is a rather simple answer and thanks in advance!

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry PI Zero

### Camera

I am using the following type of camera: Raspberry Pi NOIR v2

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: WI-FI

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

na

### Log Files

I consider the following log files relevant to this issue:

na

starbasessd, 2021-03-02:
If you have a command to turn on the LED you can do it here:
https://github.com/ccrisan/motioneyeos/wiki/Tweaks#adding-custom-startup-commands
Decent instructions for setting up a IR LED:
https://projects.raspberrypi.org/en/projects/infrared-bird-box/5

starbasessd, 2021-05-28:
No response

closed 2021-05-28 by starbasessd