motioneye-project/motioneyeos #2689: Can't view files stored on Network Share via MotionEyeOS interface

CJTheBrave, 2021-03-05:
Hi

I have been running MotionEyeOS for some time, and recently added a 'BirdBox' camera to my collection. At that time I decided to offload the file storagae to my NAS to avoid having to check free storage so often and minimise the risk of SD card failure due to years of overwriting.
The share works fine, the files get uploaded and are accessible from the network via a share on a desktop or a remote share via smartphone etc. but this means accessing 2 systems (camera to check things over, share to see if anything has been recorded).
However, I was expecting (perhaps wrongly) that the MotionEyeOS installation would still be able to see, and present, these files to me via the camera interface - i.e. via the "open movies browser" and "open pictures browser" buttons when stored on a network share.
When I click these now I get the message "Pictures taken by BirdBoxCam (no media files)" for example.
Am I incorrect in thinking that the OS should be able to see and deliver the files via this interface in the same way as it would from a local SD or USB drive ? And if my scenario is incorrect, what might I need to do to rectify it?
I can't find any reference to this anywhere to be honest.

Many Thanks in anticipation

Chris

motionEyeOS version:20200606
Raspberry Pi Zero W
MMAL Camera
Raspberry Pi NOIR camera with IR light
motionEyeOS / Pi is connected to the network via WiFi
Using a Synology NAS Box share with SMB2.0 on MotionEyeOS to remotely store the resulting videos and stills.

starbasessd, 2021-03-05:
You cannot access the videos or snapshots on the NAS from the motionEyeOS interface, unfortunately, just any local files. I have tagged this as a 'Feature enhancement' request.

CJTheBrave, 2021-03-05:
Ah, thank you starbasessd, was hoping it wasn't my configuration as everything else has worked so smoothly.
Thanks for the response and tags, let's see if it's possible. Meanwhile, back to checking both locations it's not like it's too much effort really.
Cheers

starbasessd, 2021-05-28:
BTW, you can access them in a limited setup. If you use SMB/CIFS, and use File Storage, Change Custom to Network (IOW don't use the Upload Services) it will prompt for connection info. You CAN access them then with the JPG and Play buttons, but they can be very slow to pull up the directories. If you want more info, just let me know.

closed 2021-05-28 by starbasessd