motioneye-project/motioneyeos #2690: NO Network Share on Storage Device - Debian 10

philou15, 2021-03-05:
Hye

I test motion eyes on RP3, it ok i have the file storage : 
STORAGE DEVIS : NETWORK and its Works
But i test on debian 10 because i have 14 cctv and i need more powerfull for work good. 

But i have install with the wiki https://github.com/ccrisan/motioneye/wiki/Install-On-Debian
it works ! but on file storage, i dont have the network storage 

What the problem ? and the solution ? 
Tks 
Sorry i m french 
[image]

starbasessd, 2021-03-05:
Where and what type of network storage do you have?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

philou15, 2021-03-06:
A NAS, SMB, but if it say : \\IP\ i need the user and passwork for acces. 

On PI, Have the option, Network storage with user and password.

Tks

starbasessd, 2021-03-06:
You can map your network drive to something like /media/nas and then use Custom path and point there.
Good set of instructions on how to map a network share (SMB/CIFS) to your machine is here:
https://linuxize.com/post/how-to-mount-cifs-windows-share-on-linux/
I prefer using an Identity file, and auto mount in /etc/fstab, myself.

philou15, 2021-03-06:
ok, i understand, i need to test, tks 
but it not easy method like the motioneyes on PI :)

Alternative easy it possible or not ? 
Tks for astuce !

starbasessd, 2021-03-06:
I am not sure what you're asking:
"Alternative easy it possible or not ?
Tks for astuce !"
MotionEyeOS has to do it all by itself. If you are running motion/motionEye on another OS, motionEye leaves the hard work to the OS.

philou15, 2021-03-07:
On pi, we have the option for network share like this 
It is easy for network storage, can we have the same option on motion eyes on debian ?
tks
[](url)
[image]

starbasessd, 2021-03-07:
With motionEye on another OS, you can connect to almost anything, much more
flexible.
With motionEyeOS, you are limited to what is built in, without being able
to set up your own connections.
It's not hard to add a network drive mapping, I pointed out just one set of
instructions, there are hundreds if not thousands of others.
Also, since you may have to add software, the way to add software is
different for Debian types, vs Red Hat types, vs ARCH types, vs all the
other flavors. The system would become HUGE to test for each OS type, then
also have the ability to add the needed packages for each OS.
No, it is not practical. motionEyeOS can offer it because it's only a
single OS, single set up, and needed packages for the supported network
mappings is there.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

philou15, 2021-03-08:
Ok tks for help :)

starbasessd, 2021-05-28:
No response.

closed 2021-05-28 by starbasessd