motioneye-project/motioneyeos #2692: Suggestion: Allow recording all cameras together into 1 video file.

Ryan5374, 2021-03-07:
I'm not sure about the logistics of this but my cameras are close together, meaning that it would be convenient to have any videos record into 1 video that shows all 3 cameras at the same time. Is this possible? Just a suggestion.

starbasessd, 2021-03-07:
I've tagged this as a enhancement / feature request.
I don't know what resources you have available (other PCs, Macs, Pis, Linux) but there are many packages out there to record a web or screen stream. You could record your main page (port 80 of motionEyeOS or 8765 of motionEye) of your primary hub.
I just installed simplescreenrecorder on a Pi and recorded some quick video for 5 cameras main screen. I don't know if it's possible in motionEyeOS, but it is doable in any of the OS installs (Debian/Ubuntu/RaspberryPiOS, etc...)

Ryan5374, 2021-03-07:
Sounds good!