motioneye-project/motioneyeos #2693: Camera not detected on raspi zero w

Devyang-SHIFT, 2021-03-08:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (20200606).

### Board Model

I am using the following board/model: (Raspberry PI Zero (W)).

### Camera

I am using the following type of camera: (Network Camera).

My camera model is: (Raspberry PI Camera Module).

### Network Connection

My motionEyeOS unit is connected to the network via: (WiFi).

Hey
i have been working with this for more than a month now but getting the same again and again. Please help

[image]

[Instructions followed](https://linustechtips.com/topic/927120-diy-wifi-cam-tutorial)

starbasessd, 2021-03-08:
Are you doing 2 Pi's? One with camera and one without acting as a hub?
If not, the PiCamera is attached directly to the PiZeroW then it should be
a mmal type camera, not network.
If you can SSH or attach a keyboard and monitor, run the following command:
vcgencmd get_camera
If it comes back with no camera attached, make sure the ribbon cable on the
Pi side has the blue or black tab away from the PiZero.
On the Camera side, it should also be away from the board
https://www.techcoil.com/blog/connect-raspberry-pi-camera-module-raspberry-pi-zero-w-official-case/

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Devyang-SHIFT, 2021-03-10:
yes, I have raspberry pie 4 as hub. I cant connect keyboard and mouse. But please tell me how to ssh so I can try it. Also the ribbon cable is attached nicely as the infrared lights start glowing on power. 
Thanks for reply

starbasessd, 2021-03-10:
SSH to the pi is easy, but because you haven't given what other computer
you are using to connect to the WebGUI from, I can't help with that. There
are MANY pages out there to give simple instructions depending on whether
you are using Windows 10, MacOS, or a flavor of Linux.
When you are trying to connect the Pi4 to the PiZero Camera, instead of
using Network Camera, and port 8081, try using the Remote MotionEye
connection, and http://[ip_address] and username admin (and password if you
added one on the PiZeroW
You can also connect directly to the PiZeroW Camera with a web browser to
the http:/[ip_address]:8081 or http://[ip_address] and confirm the PiZeroW
is working correctly with a video stream.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Devyang-SHIFT, 2021-03-11:
i am using a windows machine. Connectivity is also an issue as it shows error message.

[image]

I think my camera is either not recognized or its not working. Is there any way I can check it? (Infrared lights turn on when powered). Is it because of less power?

starbasessd, 2021-03-11:
You should be able to type that URL into a browser and see the camera
stream from the PiZeroW

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Devyang-SHIFT, 2021-03-11:
Nope.
Camera stream not showing up.

[image]

starbasessd, 2021-03-11:
Step by step:Find the Raspberry Pi Zero W IP address
open a browser like Edge, FireFox, Chrome, etc
Enter: 
http://[ip_addr_of PiZeroW]
or
http://[ip_addr_of PiZeroW]:8081
What is your result?
I get (screenshot) 
http://[ip_addr_of PiZeroW]
[image]

http://[ip_addr_of PiZeroW]:8081
[image]

Devyang-SHIFT, 2021-03-16:
I have tried it but did nothing. It shows this

[image]

I have added everything correctly.

Devyang-SHIFT, 2021-03-21:
please help

starbasessd, 2021-03-21:
Something is not set up correctly.
Switch user to admin
go to settings, and screen shot the Video Setting section on the hub.
In the PiZero log in as admin in the GUI, and screen shot both Video
Settings and Video Streaming

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-05-28:
No response

closed 2021-05-28 by starbasessd