motioneye-project/motioneyeos #2694: Could not change the motioneye.conf file

johnrogerpap, 2021-03-08:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: ().

### Board Model

I am using the following board/model: ( Raspberry PI 3B+).

### Camera

I am using the following type of camera: ( Network Camera).

My camera model is: (insert your camera model(Grandstream GSC).

### Network Connection

My motionEyeOS unit is connected to the network via: (Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 
  ( External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

 Hello
I know that issue is closed but i am having trouble changing the motioneye.conf file to rewritable. I am not so familiar with Linux OS and i do not know where i should enter the commands mount-o remount,rw / etc. I tried firstly on the terminal in Raspberry with no success. Then i inserted the SD card to another Linux system and i tried in the terminal over there but still the file cannot be changed to read and write. I want to access this file because an error pops up in the UI that says unable to get local issuer certificate. As far as i understand this can be fixed by adding a line in the etc/motioneye.conf file which will say validate_certs false. The funny thing is that i already have domain name and ssl certificate but i do not know how to insert them in my camera. Anyway i really want to test this OS for my IP cameras and i appreciate the work done from ccrisan. Thanks for the help

starbasessd, 2021-03-08:
Turn on Debug Mode in Expert Settings. Be sure to turn it back off after completing changes.

johnrogerpap, 2021-03-08:
@starbasessd Where is that setting? In the UI?

starbasessd, 2021-03-08:
Yes, in Settings, Expert Settings

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

johnrogerpap, 2021-03-08:
I have changed that and i tried inserting the card into the other Linux machine. When i try to open the SD card to see the files it says Error mounting /dev/mmcblk0p3 at /media/*user*/*partition*/ cante read superblock. What am i doing wrong?

starbasessd, 2021-03-08:
That is the /data partition
You need to edit the mmcblk0p2 partition, but you don't need to remove the
SDCard with Debug On, you can edit in place. SSH to it or attach a keyboard
& monitor.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

johnrogerpap, 2021-03-08:
So to edit in place i do have to execute mount -o remount,rw / or i have to do something else. Sorry for the noobish questions that i ask.

starbasessd, 2021-03-08:
Turning on Debug Mode effectively runs the mount -o remount, rw / It also
forces a reboot.
Then when you are done you turn it off, and it sets it back to read-only.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

johnrogerpap, 2021-03-08:
Ok sorry if i got this wrong. I am on the terminal in my raspberry. The command line says [root@meye-5275af85 "] where do i go from there? How can i find the file motioneye.conf?

starbasessd, 2021-03-08:
If you enabled Debugging, and have rebooted, you can edit
nano /etc/motioneye.conf

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

johnrogerpap, 2021-03-08:
Ok yes that did the work. I have changed the motioneye.conf file. But still when i put the IP adress of my IP camera on the URL field i still have the same error message  

SSL certificate problem: unable to get local issuer certificate | ?
-- | --
 

I thought that by adding the line validate_certs false in the motioneye.conf would fix the problem.

starbasessd, 2021-03-08:
What camera make & model and URL are you using?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-08:
BTW, that validate_certs solution was for motioneye, not motioneyeos. Yes,
there's a difference.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

johnrogerpap, 2021-03-08:
Camera model GSC3615
URL is https://192.168.1.3/index.html?t=1615268356926

starbasessd, 2021-03-08:
> Camera model Grandstream GSC3615

try
rtsp://192.168.1.3/0
for your camera

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

johnrogerpap, 2021-03-08:
Wow man thanks!!!!!! That did the trick. Awesome!!!!! Thank you so much for everything!!

starbasessd, 2021-03-08:
Info came from:
Connecting iSpy to IP Cameras <https://www.ispyconnect.com/sources.aspx>
They have a massive database on camera URLs.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-03-08 by starbasessd