motioneye-project/motioneyeos #2695: Accessing Pictures on my SD card

dermotx, 2021-03-08:
Hi,
I'm new to this and have limited Linux experience. I have searched this forum for the solution to this issue and the wider internet also. I have motionEye Version | 0.42.1  Motion Version | 4.2.2+gitUNKNOWN OS Version | motionEyeOS 20200606. I am using 3 raspberry pi 3 with the official raspberry pi cameras, not NOIR.

I am using the software in the Still Images Mode only but can't access them over the local network because there are too many pictures. I have the setting as taking a picture every 3 seconds on each camera. 

Is it possible to access the pictures directly from the SD card, preferably on a win10 laptop? If not how could I access the pictures using the raspberry pi's?

Thanks
Dermot

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2021-03-08:
You can turn on FTP and access via wsftp.
You can turn on SMB and access it via Windows File Sharing
Both are found under Settings, Services.
If you have ssh access, you can access via scp (PuTTY has scp, many other
versions...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dermotx, 2021-03-08:
Thanks for that starbasesd. The FTP option worked great.
Regards
Dermot

closed 2021-03-08 by dermotx