motioneye-project/motioneyeos #2696: Folder created, but no images stored in Google Drive

mrglobetrotter0088, 2021-03-09:
Hello everyone!

I have built a bird box camera system with a raspberry pi, pijuice and the motioneye software. The idea was, that the raspberry pi boots up every morning at 7 a.m. and snaps every minute 1 image of the box. After 15 minutes the raspberry pi shuts down (sudo halt command). The system is powered by pijuice with an attached battery and a solar panel.

During testing everything worked correctly and the 15 images were always uploaded. However, I have used the electricity coming from the socket instead from the solar panel. Now the bird box is set up and on the first day no images were uploaded. 

I am wondering if the problem occurred due to a lack of electricity or wifi signal in my backyard. The folder for the uploading images was created correctly but no images were stored. Also another script (telling me the status of the battery) was uploaded correctly to Dropbox. And during that time, the whole system was powered by the battery. Unfortunately, I cannot supply you with any motioneye log-files.

What are your suggestions for this issue? Is it more on the wifi signal or on the insufficient solar panel side causing motioneye failing in uploading images to google drive? I would have the possibility to reinforce the wifi signal in that spot of my garden and also to attach the raspberry pi to a more stable electricity source. But I am interested in your thoughts.

Thank you very much for your help and comments!

starbasessd, 2021-03-13:
First, you don't give much pertinent info:
make/model of board
which version of motionEyeOS
Are you able to ssh to the board when in place?
Are you able to ping the board in place?
Bandwidth testing on an iphone has nothing to do with the issue. Wifi usage on the iPhone tells you only that. The board uses a different chipset, antenna arrangement, etc.
You need to get logs from the Pi, or able to run the command:
vcgencmd get_throttled
and post the results. This will let you know if the power and temerature of the board (assuming a broadcom based board and OS supports vcgencmd, like PI, are stable and appropriate.

mrglobetrotter0088, 2021-03-14:
Thank you for your reply. 

First of all, I have to inform you that after my previous post it worked all fine. What I have done is just changed the power source from the solar panel to the electricity socket with a 5m long USB cable. The pictures were uploaded correctly. After three days I got the same issue again. However, the text file was uploaded to Dropbox without problems but the images are missing. Therefore, I think it is not a power or wifi problem. Might it be because of the storage on the Raspberry Pi? Maybe once the images are deleted on the disk (32 GB micro SD), the uploading works again fine?

Here the additional info.
- Raspberry Pi Zero W, with a PiJuice Zero attached.
- I think I've installed the newest one, version 0.42.1 or so? I'm not sure.
- I have to try to ssh or ping it during the operation time, since it is only 15 minutes during the day "awake", the rest of the time it in in "sudo halt".

But maybe someone experienced similar phenomena, as described it the first section.

starbasessd, 2021-03-14:
I saw that it worked on the bench, but you say it isn't working on the box.
'the newest one'? What is the date on the img file? 20190911, 20200203, 20200606, dev20201028?
In order to help figure out what the issue is, we need the other information. 
I have not seen anyone else with a PiJuice on here. Similar issues usually end up pointing to power issues, or heat issues, and the Pi throttling (shifting to low power mode to protect itself) and networking is among the first things to go wonky. That is why I asked for the additional information.
32GB data cards should be more than enough space for a while (not just a day or 3)
If you are using a soft shutdown (Like halt) it should be able to complete tasks prior to shutting down, and not corrupting data.
How did you plan for maintenance? Extra time one day a week? Pull it back down?
It comes down to a couple possibilities, off the top of my head.
Power issues, low voltage, etc. 
Thermal issues, the thing can't cool itself.
Wifi strength issues (and yes, I saw your comment about the iPhone doesn't have issues, but it's not a Pi)

closed 2021-03-31 by mrglobetrotter0088