motioneye-project/motioneyeos #2697: script turns off camera while I'm home

Altdorf25, 2021-03-11:
Hello,

I am using MotionEye with a Raspberry Pi 3B and a Microsoft LifeCam HD-5000 camera. 

Is there a feature in MotionEye or a script to turn off the camera if someone in the household is home and turn the camera back on when everyone is gone?

I would like to use the detection of the cell phone IP address of family members.

I am new to Linux and am unable to write this script myself.

starbasessd, 2021-03-11:
Duplicate of post over on motioneye github
There are a number of variables here:
How many to check for?
How often to check?
Are they set up with static IPs?
How would you 'guarantee' that they are the 'real' phones of the residents? (not someone 'spoofing'

Is the system capable? Yes
Is there a script available? Not that I am aware of.

Altdorf25, 2021-03-12:
yes, suddenly we continue here 
https://groups.google.com/g/motioneye/c/3b4QDHY1LUI

starbasessd, 2021-03-12:
There is Working Schedule, which would control by time of day.
If you are not at home 8a to 5p Mon - Fri, you could have it on recording during that time, but off the rest of the day, and a different schedule on Sat & Sun.
As to the Location, you would need to find someone willing to write the scripts.