motioneye-project/motioneyeos #2698: HDD does not show up in drop down menu

johnrogerpap, 2021-03-15:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (20200606).

### Board Model

I am using the following board/model: ( Raspberry PI 3B+).

### Camera

I am using the following type of camera: (various cameras).

My camera model is: (various cameras).

### Network Connection

My motionEyeOS unit is connected to the network via: (Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/6144051/dmesg.log)

Hello. New issue here :) I have an external HDD which i think i formatted as ext4 following those instructions 
https://github.com/ccrisan/motioneye/issues/820
https://kwilson.io/blog/format-a-linux-disk-as-ext4-from-the-command-line/
But still the HDD does not show up on the drop down menu on the GUI. What am i doing wrong? Thank you in advance
If any other log file is needed just let me know

starbasessd, 2021-03-15:
A couple of things:
Your Pi is undervolting due to power drawn from the PI, in the log
I would bet that you have a 500GB Buffalo external. It is finding it, but it's not formatted correctly or corrupt. 
10.931991] scsi 0:0:0:0: Direct-Access     BUFFALO  HD-WLU3/R1     1 0001 PQ: 0 ANSI: 6
[   10.932775] sd 0:0:0:0: [sda] 488376678 512-byte logical blocks: (250 GB/233 GiB)
[   10.933055] sd 0:0:0:0: [sda] Write Protect is off
[   10.933062] sd 0:0:0:0: [sda] Mode Sense: 1f 00 00 08
[   10.933351] sd 0:0:0:0: [sda] No Caching mode page found
[   10.933357] sd 0:0:0:0: [sda] Assuming drive cache: write through
[   10.966543]  sda: sda1
[   10.966563] sda: p1 size 488392704 extends beyond EOD, enabling native capacity
[   10.968049]  sda: sda1
[   10.968055] sda: p1 size 488392704 extends beyond EOD, truncated
[   10.969617] sd 0:0:0:0: [sda] Attached SCSI disk
It's not seeing it correctly.
What do you get if you type in the command 
df -h
???

johnrogerpap, 2021-03-15:
> A couple of things:
> Your Pi is undervolting due to power drawn from the PI, in the log
> I would bet that you have a 500GB Buffalo external. It is finding it, but it's not formatted correctly or [...]

I got this
[image]
Is it a problem to be undervolted?

starbasessd, 2021-03-15:
Yes, it means the power supply being used is being stressed, and can't
supply enough power. If you don't leave your mouse & keyboard on it all the
time, you should be OK for now.
You want to pick Custom Path, and type in /data/media/sda for the path.
That will give you 228 GB of space available.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

johnrogerpap, 2021-03-15:
Oh ok i thought that it would show up in the dropdown menu.

johnrogerpap, 2021-03-15:
Thank you by the way!!

starbasessd, 2021-03-15:
No prob. Close the issue if its resolved.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-03-15 by johnrogerpap