motioneye-project/motioneyeos #2699: Motioneyeos problem showing console on pc

ted8862, 2021-03-15:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026 Oct 25, 2020

### Board Model

I am using the following board/model: Raspberry Pi 4 Model B with 4gb

### Camera

I am using the following type of camera: Raspberry Pi camera  V2 with 8mp

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

I have tried to use releases 20200606 and 20200203, but could not get Motioneyeos to boot without inserting the "known good files" fixup4.dat and start4.elf. The boot worked well after that, but my camera would not connect no mater what I tried. So finally I booted with pre-release dev20201026. This came up on my pc as expected and the camera worked! But when I logged in, the motioneyeos console was barely visible under a grey screen. The camera picture looked great and was streaming. But I could not access any of the console tools. Any help would be greatly appreciated.

starbasessd, 2021-03-15:
In motionEyeOS, the console is command line only. The WebGUI is accessible from another computer or phone on the network.
Also, motionEyeOS is locked down, with no ability to use apt-get utilities. You can add python scripts and such in the /data folder tree.

ted8862, 2021-03-15:
Thank you for your very quick response. I should have mentioned that my problem is with the WebGui. I use the rpi board ip address to log into motioneyeos. I have used your advice from previous comments to run a complete firmware and software update and see that vcgencmd bootloader is a new version. The versions 20200606 and 20200203 would boot normally with the start4.elf and fixup4.dat added but would not connect to my pi camera. The WebGui looked fine and I tried a number of settings to access the camera without success. The version dev20201026 was the first time I could boot normally without changing any files and the camera came up streaming looking great, but the WebGui was grey and I could not access it. ??

starbasessd, 2021-03-15:
Have you cleared cache and cookies in your browser?
Have you clicked on the icon next to the settings icon, and logged in as
admin? Sometimes it loads you as user, instead by default.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-15:
Also, if you can post a screenshot of the issue in case we aren't on the same page?

ted8862, 2021-03-16:
[image]
[image]
The me01 jpeg is a screenshot of my access to the WebGui of motioneyeos if I do not login but do press the top left button.
The me02 jpeg is a screenshot of the WebGui if I try to login with admin but no password.
My guess is the WebGui is expecting a password but none has been set up.

starbasessd, 2021-03-16:
I've seen that before, the sdcard was corrupted. I would reimage.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ted8862, 2021-03-16:
First of all, I want you to know that I am very grateful to you for taking the time to assist me. I am far from an expert on HTML, but when I looked at the HTML generated by the WebGui, I noticed that errors at Login stated code was not compatible with Chrome, Edge, Firefox and others. So I hooked up another rpi to run the WebGui and everything worked! I had tried this before on other versions of Motioneyeos and it did not work. So for me the pre-release software was the answer. Thank you again.

starbasessd, 2021-03-16:
I use FireFox 86, Edge 89, Chrome 89, and IE11 from Win10 21h1, all work OK.
I use several browsers on RaspberryPiOS, including FF and Chrome.
There have been issues in the past with specific browsers and specific
versions of motionEye & motionEyeOS, but AFAIK, none currently.
If you want to look into this further, let me know, otherwise, if you are
working, please close.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-03-16 by ted8862

bryanbarnes01, 2021-03-19:
I had a similar issue - I have 2 different zero's, same network, webGUI. One had Motion Ver. 4.2.2 (same OS name dev20201026). It had all the admin features - add camera, Movie, Motion etc . . . Second camera had everything the same. Updated one to Motion Ver 4.3.1, admin features disappeared. Updated second to 4.3.1 - same thing. Downgraded (reinstalled ) to Motion Ver. 4.2.2 on both Zero's to be able to use those admin features via WebGUI. Did this today.

starbasessd, 2021-03-19:
Curiosity, Bryan Barnes:
How did you Upgrade motion? dpkg and apt are not installed in motionEyeOS.
Latest motion package is 4.3.2-1
https://github.com/Motion-Project/motion/releases
There isn't any ability to compile from source in motionEyeOS.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

bryanbarnes01, 2021-03-19:
There is a check software update button. Whats odd is this versions WebGUI doesnt provide the same admin functionality as the earlier version. Add camera, motion, filie, movie setting are all missing.Even weirder is I will reflash the earlier version, it boots up fine, add camera, have all the admin funcionlity then reboot log in as admin with the functionality missing again - strange. The only thing i am doing diferent is going from USB ethernet, then adding the Wifi info in the WebGUI (wpa_supplicant didnt work for me), then rebooting to the Wifi.

Bryan Barnes | MA, CTT+
(c) 214.810.6086 

 
 

https://github.com/Motion-Project/motion/releases

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-19:
Install 20200606 to 8GB SD Card, add wpa_supplicant.conf, ssh.txt
Boot in PiZeroW
loads normally, normal WebGUI, add camera, modify settings, no issues.
reboot
No issues.
Shows
motionEye Version 0.42.1
Motion Version 4.2.2+gitUnknown
OS Version motionEyeOS 20200606
Click on Software Update Check
motionEye is up to date (current version: 20200606)
Expert Settings: Enable prereleases Click to ON
Click Apply
General Settings, Software Update click Check
New Version available: 20201026. Update?
Click Yes
Shows
motionEye Version 0.42.1
Motion Version 4.3.1
OS Version motionEyeOS dev20201026
Run same tests, all act normally, all admin functions available
[image]

starbasessd, 2021-03-19:
If I log on as user with user password, I get the following screen. If I log in with admin and admin password, I get the full menu options, with both versions.

[image]

ted8862, 2021-03-19:
I tried several versions of motioneyeos (203 & 606), but only pre-release 1026 worked for me and it does not have any problems. However, I wanted to share one tip if you want to use wifi instead of ethernet. You will need to create a text file - wpa_supplicant.conf. I used Notepad++ and it will add .txt to the file. You will not be able to delete the .txt in windows normally. You must go to the command line and enter: "move wpa_supplicant.conf.txt wpa_supplicant.conf" without the quotation marks. After you flash your sd card, then paste wpa_supplicant.conf before you boot for the first time.

starbasessd, 2021-03-19:
In Windows, Open File Explorer, click on the View tab, far right place check marks in File name extensions and Hidden items. You can now change file name extensions from .conf.txt to .conf easily. And when you do a file, save as, and you type in wpa_supplicant.conf it won't auto-add .txt, either.

bryanbarnes01, 2021-03-20:
@starbasessd - Success! thank you your info was spot on - got it woking with the pre-release option checked, got it updated, all admin option still visibe via the webGUI - connecting via USB ethernet, no issues.Rebooted and added wpa (note: I am having user issues with getting the wpa_supplicant_conf onto the root after flashing - sd card becomes unusable after flashing in windows - I will work on this ***@***.*** - i will try the cmd move to get the supplicant onto the newly flashed sd card later - thanks for the tip!!
oh and one other thing I was doing wrong the whole time:Expert Settings>Fast Camera - rebootThis was the reason why I was 'loosing' the admin options - camera is the generic v1.3 camera.Camera Type was the key (MMAL Camera (video core) frames are bad)I went back and forth adding camera 1) MMAL Camera (bad frame rates) to 2) Expert Setting > Fast Camera - rebooting makes the camera show up as MJPEG with no fancy admin option.

Bryan Barnes | MA, CTT+
(c) 214.810.6086

starbasessd, 2021-03-20:
Oops, yeah, even I keep forgetting that option, so few use it and it really is a great option, especially for PiZeros.