motioneye-project/motioneyeos #2700: SOLVED: wpa_supplicant.conf & brcmfmac issue on setup

Normanras, 2021-03-16:
- [ X ] I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

- [ X ] I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

MotionEyeOS version: 20200606.
Board/model: Raspberry Pi ZeroW
Camera: RPi NOIR Camera V2
Camera model: 8mp Sony IMX219 (This was bought as a bundle from Adafruit. [See this listing](https://www.adafruit.com/product/3415).)
Network: Wifi via Verizon Fios.
No log files or peripherals to note.

Setup: Balena Etcher on M1 Mac running 11.2.1 Big Sur.

---------------------------------------------------------------
 
# Command Line Solution to First Boot Wifi Issues (brcmfmac)

I was having a ton of issues setting up and getting the wifi to work on a new Rpi Zero W with a clean install of MotionEyeOS using Balena Etcher. I checked #2254, #2537, #2060, and #1604 among many others. 

2060 and 1604 were important because of the tip regarding the EOL conversion to Unix. Unfortunately, I don't have access to Notepad++ (I'm on a Mac and my PC is out of commission for the moment) and Atom's  line-end-selector package didn't successfully convert the file. This isn't my first raspberry pi rodeo and I've always had success with a headless setup using Mac's native text edit file. Simply creating the wpa_supplicant file and changing the extension to .conf has always worked for me. There is something in MotionEyeOS that didn't allow this to work.

I got it to work by going through the command line to add and edit the file. Here are the steps:

1. Always best to start with a fresh install/flash. After Balena Etcher flashes the OS, it will unmount the SD card. Remount it through disk utility (select the SD card and click "Mount" in the toolbar at the top. You can also eject it and put it back into your card reader. Remember the name of your drive. For this example, mine is called "NO_NAME".  
2. Open your terminal (if on a Mac) or command prompt (if on Windows). 
3. Type "nano /Volumes/NO_NAME/wpa_supplicant.conf"
4. You will be presented with the ability to edit a new file. Type in all the normal wpa_supplicant.conf requirements. I _did not_ copy and paste to ensure I didn't have any errors. Make sure you keep the quotation marks for SSID and PSK. Here is what you need to type:
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country= your_ISO-3166-1_two-letter_country_code
   
network={
    ssid="your_wifi_ssid"
    psk="your_wifi_password"
    key_mgmt=WPA-PSK
} 
```

5. NOTE: While in terminal and editing the file, I hit tab on the new line for SSID. I'm not sure if indentation matters, but it placed "ssid=" almost under the curly bracket after network as opposed to under the t in the example above. I would revisit this step if need be. 
6. Exit the file (Ctrl+x). It will ask "Save Modified buffer?" Choose Yes (Y). 
6. It will then ask you: "File Name to write:" Ensure you write the entire file path _correctly_. In this case it is: `Volume/NO_NAME/wpa_supplicant.conf` and hit enter.
7. It will bring you back to your command line. Go into your finder or other file system and open the SD card. Ensure the file is there. 
8. Eject it, plug it into your RaspberryPi Zero W and if you're monitoring the startup processes, watch if it connects! And if it doesn't connect, I would try a fresh flash of the SD card and revisit steps 4 and 5. 

I think this method works because there is no confusing the computer/OS as to what file type is being saved. There may be something with TextEdit on a Mac and needing to convert the file extension from .rtf or .txt to .conf and it doesn't fully apply to the code in the file. Who knows. But please let me know if this worked for you and if you have any questions!

poweribo, 2021-03-16:
#3 says Volumes but #7 says Volume. Which one is it?

NeutronKen, 2021-03-26:
I am going through this now.  I can't seem to get anything to work.  However, I can run Raspbian and manipulate the camera.  I can't seem to get the motioneyeOS to work even with these instructions.

starbasessd, 2021-03-26:
Which Pi do you have?
Which version of motionEyeOS are you using?
Which program are you using to image the SDCard?
Can you connect a monitor and keyboard to the Pi?
What does it show?

gyurobenjamin, 2021-04-30:
Didn't do the trick for me

starbasessd, 2021-04-30:
@gyurobenjamin 
What didn't do the trick for you? What is your setup?

Normanras, 2021-05-01:
@NeutronKen - jumping off of what @starbasessd asked: what do you mean you can manipulate the camera? Do you mean by using raspivid or raspicam? Please ensure when you're typing things in the command line that there are no errors. Including, using quotations for your SSID and PSK.

starbasessd, 2021-05-28:
No Response from any poster

closed 2021-05-28 by starbasessd

bbillp, 2021-05-29:
Why are there so many emails today? 30-40 received.

starbasessd, 2021-05-29:
If you are Subscribed to the whole project, I was cleaning up some of the
stale issues.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

anderson-eric, 2021-07-04:
I too have tried the different options to get eyemotion to fix the brcmfmac issue. Eventually I verified my wpa_supplicant.conf file using it with a different image file (2021-05-07-raspios-busterarmhf-lite.img). I was able to connect to my wi-fi using that other image, so I knew my supplicant file was correct. 

I was ready to give up but then I tried the writeimage.sh script file listed on the installation page. For some reason the script configured the wireless correctly and the Pi Zero connected to the wireless network correctly.

For the record, I'm using WPA2-PSK and connecting to my 2.4GHz network, not my 5.0Ghz network.

The wpa_supplicant.conf file I am using:
`country=us
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
	ssid="2.4GHz ssid-name"
	prot=RSN
	key_mgmt=WPA-PSK
	pairwise=CCMP TKIP
	group=CCMP TKIP
	auth_alg=OPEN
	psk="wireless-password"
}`

dudeitssm, 2021-09-07:
> I was ready to give up but then I tried the writeimage.sh script file listed on the installation page. For some reason the script configured the wireless correctly and the Pi Zero connected to the wireless network correctly.

I just had the same experience. I shrunk the original script down just to what I need, that is, just the part responsible for making the `wpa_supplicant.conf` file. I copied and pasted it in the 32MB partition and then it worked.

Relevant part that makes the conf file below. Replace the SSID and PSK fields, then run it as a regular user, then copy the output file to the 32MB partition:
```
#!/bin/bash

SSID="mywifi"
PSK="mywifipassword"

function msg() {
    echo " * $1"
}

if [ -n "$SSID" ]; then
    msg "creating wireless configuration"
    conf=./wpa_supplicant.conf
    {
    echo "update_config=1"
    echo "ctrl_interface=/var/run/wpa_supplicant"
    echo "network={"
    echo "    scan_ssid=1"
    echo "    ssid=\"$SSID\""
    if [ -n "$PSK" ]; then
        echo "    psk=\"$PSK\""
    fi
    echo "}"
    echo ""
    } > "$conf"
fi
```

bazaboyo, 2021-09-19:
Hi
I was having exactly the same issues,  I spent many hours trying, but kept getting the error messageI, so I followed "Normanras commented on 16 Mar (above)" I was using an Apple iMac with a Raspberry Pi Zero W, I followed each step very closely and it worked for me. I downloaded the latest version of Motion eye, and every time it didn't go quite right, I re-installed a clean Motion Eye, I used ApplePiBaker to do this, if you have a problem with the wpa_supplicant.conf file moving onto your SSD double check the path, and spelling word for word. All working fine.  Many thanks Normanras .