motioneye-project/motioneyeos #2701: Inhibit Reboot after stopping server service

Gnupf, 2021-03-18:
I have problem that I hope someone can help me with.
I'm running MotionEyeOS on a Pi Zero W on battery that is being charged by solar cells.
In order to save power during the night, I want to reduce power consumption as far as possible.
I ran the command 

meyectl stopserver -b -c ~/../etc/motioneye.conf

which successfully stops the service and cuts the prw consumption by half.
However, after a few minutes, the Pi reboots by himself and the service starts again.
Is this a safety measure within the OS?
Can I inhibit this feature somehow?

Best regards
Gnupf

starbasessd, 2021-03-18:
It's part of the watchdog service, to keep the system running in the event of a software failure. 
https://github.com/ccrisan/thingos/wiki/watch.conf
I am not sure that is an option, or if there is a different format for that option, as the documentation hasn't been updated:
https://github.com/ccrisan/thingos/wiki/watch.conf
even though @ccrisan suggested it...

Gnupf, 2021-03-22:
Thank you very much for your help!

Eventually I could solve the issue by changing the "meyewatch_disable" entry to "true" in the etc/init.d/S85motioneye file.

Now I'm able to disable the server service via meyectl and also to disable the Wifi Tx to save even more power.

closed 2021-03-22 by Gnupf