motioneye-project/motioneyeos #2703: GH-Pages / motioneye.org seems taken over by spammer

LarsKumbier, 2021-03-19:
The currently linked [GH-Pages environment](https://github.com/ccrisan/motioneyeos/deployments) does redirect to a crypto-spam-page or just shows a blank page. The page should be removed from the deployments and any documentation, if it can't be regained via other means.

starbasessd, 2021-03-19:
The valid github home pages are
motioneye: https://github.com/ccrisan/motioneye
motioneyeos: https://github.com/ccrisan/motioneyeos
Where is there any link to that page?

LarsKumbier, 2021-03-19:
It's linked in the environments:
[image]

and then linked in the deployments:
[image]

starbasessd, 2021-03-19:
Thank you. Tagged and notified.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?