motioneye-project/motioneyeos #2704: Is it possible to install NGINX reverse proxy on RP with MEOS to get SSL certificate?

jedtrently, 2021-03-21:
I keep running into dead ends with many many google searches. It seems you might have to put the reverse proxy on a seperate machine or use ME?

starbasessd, 2021-03-21:
That is correct. motionEyeOS is locked down to where you cannot install other applications, such as NGINX in it. To do what you are requesting, you'd have to install a base OS (Like RaspberryPiOS Lite, or Ubuntu 20.04 server or similar, then install motion/motionEye and follow the instructions here:
https://github.com/ccrisan/motioneye/wiki/Running-Behind-Nginx

jedtrently, 2021-03-21:
Thanks for clearing that up! Appreciate it. Any idea how I might get an SSL cert whilst running MEOS? Or is this not possible also?

starbasessd, 2021-03-21:
The web service currently installed in motionEyeOS does not support https: / port 443 traffic, AFAIK, and there isn't any documentation for enabling it. You could rebuild it and add it yourself, there is documentation or building from source here:
https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source
but not specifically adding the web services...

starbasessd, 2021-05-28:
No response

closed 2021-05-28 by starbasessd

Mega1024, 2022-01-04:
That's.. kind of annoying. I know its offered for free and I've no basis to complain (and its pretty awesome as it is) but it would be nice to use MEOS, and have an https proxy on the same physical rpi. Having to setup up an entirely separate rpi to run the proxy on seems ridiculous.

starbasessd, 2022-01-04:
Works fine if you install another OS, like RaspberryPiOS (Buster or Bullseye) motionEye, and nginx on the same Pi. You get most of the same functionality (minus Fast Network Camera) and can install other functions like NFS support, nginx, etc...

starbasessd, 2022-01-04:
motionEye install instructions are here:
https://github.com/ccrisan/motioneye/wiki/Installation
In the right hand column are OSs that have been tested to be 'known working'. If it's a Debian derivative (Mint, RaspiOS, Ubuntu, etc) they are all easy installs. I even got it to work in Arch...

Mega1024, 2022-01-04:
Oh I am aware it can be installed on a base OS, but having the existing sd card image ready to go is a massive time saver.

It would be nice if https functionality could be added to the MEOS image itself (with the option to enable it, or not) Maybe ccrisan will consider that someday.

starbasessd, 2022-01-04:
I tagged for enhancement/feature, it'd be nice if someone would write up the mods needed and submit as a PR...