motioneye-project/motioneyeos #2705: clean video output from motioneyeos 2020 on RPI3

jojo2203, 2021-03-21:
Hello,

i would like to have a clean video, i.e. without border, control characters, header for further processing in obs studio. At the same time, however, I would like to have the action control buttons displayed on the mobile phone or a second computer in order to be able to control the pan tilt. How could I implement this? 

Thanks. jojo

starbasessd, 2021-03-21:
Turn off text overlay, and feed from port 8081, however you lose the text overlay. If its a network cam, you can load the camera as camera 1 and camera 2 and feed from port 8081, and  run from camera 2

starbasessd, 2021-03-21:
BTW, it would be helpful to know the info requested in the issue template to help more fully.

jojo2203, 2021-03-24:
Thanks for the fast help. The additional specification of the port number 8081 in OBS Studio was the crucial point. Now I have a clean image even without the header.

Thanks. jojo

closed 2021-03-24 by jojo2203