motioneye-project/motioneyeos #2708: unable-to-open-video-device / 4l2_device_capability

adinovak0402, 2021-03-25:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: **motioneyeos-raspberrypi4-20200606 (4).img.xz**

### Board Model

I am using the following board/model: **Raspberry Pi 4 Model B - 2GB RAM**

### Camera

I am using the following type of camera: (choose from V4L2, MMAL). - **RPi Camera (H)**

My camera model is: **RPi Camera (H)**

### Network Connection

My motionEyeOS unit is connected to the network via: **(Ethernet)**.

### Peripherals

### Log Files

I consider the following log files relevant to this issue:

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/6206540/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/6206541/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/6206542/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/6206543/motioneye.log)

I must also point out that the camera worked for me in normal raspbian, please for any help / advice.

starbasessd, 2021-03-25:
1st, thanks for bringing it over here.
I am going to assume the camera is a CSI Pi Camera, please correct me if incorrect.
It should auto-set up as a MMAL type camera.
If you ssh or console to the Pi4-2GB, can you run
vcgencmd get_camera
and post results here?
Can you post 
vcgencmd bootloader_version
results here?
Can you try motionEyeOS dev20201026
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz
there are a lot of issues with 20200606 and RPi4s.

adinovak0402, 2021-03-25:
I honestly don't know what CSI is, but I bought it here:
https://rpishop.cz/kamery/1045-waveshare-rpi-kamera-h.html
[image]

starbasessd, 2021-03-25:
It's the ribbon cable connection.
It's not showing up, maybe because the cable isn't connected correctly.
Please check that the cable is installed correctly.
https://www.youtube.com/watch?v=lAbpDRy-gc0

adinovak0402, 2021-03-25:
I'm really sure the wiring is ok, I tried to buy a new cable, but without success, I really don't know what the mistake is here, because when I run normal raspbian the camera worked normally. Here even if it changes the camera to any other on it does not work.
[image]

starbasessd, 2021-03-25:
If it works with RaspberryPiOS as it sits, the I'd suggest using the dev20201026 image for the Pi4.

adinovak0402, 2021-03-25:
It works! Thank you so much for the advice, I know I did well that I wrote here, for others who have the same problem so this img works https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4 -dev20201026.img.xz

closed 2021-03-25 by adinovak0402