motioneye-project/motioneyeos #2709: Network Share Works but doesn't

ShosMeister, 2021-03-27:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20200606.

### Board Model

I am using the following board/model: RPi Zero (2) and RPi 3B+.

### Camera

I am using the following type of camera: MMAL

My camera model is: RPi v2.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * USB External Hard Disk)

### Log Files

Share is setup on the hub
[image]

Have 2 RPi Zeros w/cameras. RPi 2 works for the network share
[image]
 
RPi 1 does not - fails setting from web UI.
[image]

Manually mounting the same point from command line (SSH) works.
[image]

Once there, I can use the custom path file storage from the web UI and it does save images there.

I don't mind if I need to have it set using custom path rather than network share but if I reboot, the mount point will be gone and have to be manually remounted before the camera will work again.

From the motioneye.log after I try to test the connection from web UI:
2021-03-27 12:52:00: [motioneye]    ERROR: failed to mount smb share "//172.20.15.220//storage/sda1/MEye-01.d" at "/data/media/motioneye_172_20_15_220__storage_sda1_meye_01_d_admin"
2021-03-27 12:52:00: [motioneye]    ERROR: access to network share //172.20.15.220//storage/sda1/MEye-01.d failed: cannot mount network share

Options? Suggestions?

starbasessd, 2021-03-27:
Rpi0-2 doesn't have the leading / for share name, Rpi0-1 does have a leading / for the share name.
which causes the error:
2021-03-27 12:52:00: [motioneye] ERROR: failed to mount smb share "//172.20.15.220//storage

ShosMeister, 2021-03-27:
Got it. Thanks!

closed 2021-03-27 by ShosMeister

starbasessd, 2021-03-27:
Glad to help. Sometimes you can't see the forest for the trees...