motioneye-project/motioneyeos #2710: Dual Network Connections

ShosMeister, 2021-03-28:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606.

### Board Model

I am using the following board/model: RPi zero(2) and RPi 3B+.

### Camera

I am using the following type of camera: MMAL.

My camera model is: RPi v2.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi AND Ethernet.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * USB External Hard Disk)

I have managed to get the system running - 2 RPi Zero's with cameras and a RPi 3B+ as the hub with attached storage. All is working fine except I want to be able to access the hub from my wired network rather than the WiFi. Obviously I need to keep the WiFi setting as that is how the three Pis communicate but for browser connectivity/monitoring, I'd like to use the eth0.

I can't setup both in the web UI and while I can ifconfig eth0 address it all day, if I reboot, that goes away?? So how can I enable both interfaces and have them both come up with a static IP at boot?

Thanks!!

starbasessd, 2021-03-28:
If the PiZeros are on the same network (via a WiFi Router or Access Point) as the Wired network, you don't need the WiFi on the Pi 3B. 
If you are wanting to use the Pi3B as an Access Point, using motionEyeOS, that precludes the Ethernet.
If the 2nd option is what you want, then I'd recommend using RaspberryPiOS and install any number of options to make it act as an Access Point/Router, (example here):
https://learn.adafruit.com/setting-up-a-raspberry-pi-as-a-wifi-access-point/install-software
then add motion/motionEye from here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian 
I personally add static IPs via the DHCP service on my router/DHCP server.

ShosMeister, 2021-03-28:
Nope. I want to have all three of them on the WiFi talking to each other. And my WiFi is isolated from my home network so not even close to the same subnet. But if I put the Pi3 on both, then I can web to it via my lan and it can still see the remote cameras on the Zeros.

ShosMeister, 2021-03-28:
Oh and yes, I have reserved the IP for the devices on my router via MAC.

starbasessd, 2021-03-28:
Then your best choice would be running RaspberryPiOS Lite on Pi3, and
adding the setup for AP/Router and motion/motionEye. You won't be able to
set up a split network on the Pi3 with motionEyeOS.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ShosMeister, 2021-03-28:
But I can add the address manually from the command line. Guessing there is a config file that is read at boot time to setup the interfaces. Or is it that the web UI can't listen on more than one interface?

starbasessd, 2021-03-28:
Not from the command line, but there is a file (or group of files) to set
up Access Point Mode, which also work for setting up a static IP address:
os.conf << used to disable need for any network configuration
static_ip.conf << sets a static IP address (only for the primary network,
eth0 or wlan0)
https://github.com/ccrisan/motioneyeos/wiki/WiFi-Access-Point
Like I said, though, if you can, you should have your DHCP server issue IPs
to specific MAC addresses.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-05-28:
No response

closed 2021-05-28 by starbasessd