motioneye-project/motioneyeos #2711: Raspberry Pi Zero W - NoIR Cam Stuck in 640x480

DustyArmstrong, 2021-03-29:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

0.42.1.

### Board Model

Raspberry Pi Zero W.

### Camera

MMAL in fast network mode.

NoIR V2.1

### Network Connection

WiFi.

I have two Raspi zero W's, both with the exact same setup, same camera, MotionOS image etc. One of them seems unable to go above 640x480 in the settings. I can select higher resolutions, but it remains in the same aspect ratio and the data throughput on my WiFi controller doesn't change for that host. 

I'm not sure if this is a physical defect of the board or camera, or an issue with MotionEyeOS. Has anyone experienced this? I imagine my best course of action is to reinstall everything and reseat the camera.

starbasessd, 2021-03-29:
Which image did you install? 20190911, 20200203, 20200606, or dev20201026?
My first option would be to re-image with either 20190911, or dev20201026, personally, and yes, re-seat the cable ends on the one not working well.
I would also consider setting the camera to 'Disabled' temporarily, and run the command
raspistill -o cam.jpg
and check the pic's resolution.
raspistill defaults to the highest resolution

DustyArmstrong, 2021-03-29:
I am running 20200606 on both of them, so it should be OK in theory. The other one is working as expected.

Running the command produced: 

```
[frontcam ~]# raspistill -o cam.jpg
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates
```

Does this provide insight, I imagine that isn't right...

DustyArmstrong, 2021-03-29:
Additionally. 

I had issues with this particular device from the get go. The camera wasn't automatically detected like it was on the other pi. There's probably an actual problem with the camera module itself.

starbasessd, 2021-03-29:
Does it work with RaspberryPiOS and an app like Cheese?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-29:
Also try the command
vcgencmd get_camera

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DustyArmstrong, 2021-03-29:
Thanks for your help starbasessd. 

I reseated the camera multiple times, and now it appears to be working. I noticed that when I would close the case, I re-opened it out of curiosity to find the camera half hanging out of its connector. I guess it must have been a sort of, "N64 cartridge tilt" bug where it was technically working, but wasn't. 

I'll close this off! Reseat your cameras if this happens to you!

DustyArmstrong, 2021-03-29:
Nevermind, changing it to fast network camera again after this caused it to revert to the broken state. I'm just going to RMA the unit.

closed 2021-03-29 by DustyArmstrong