motioneye-project/motioneyeos #2714: Connection timeout after some time (random: minutes to hours)

arielylim, 2021-04-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry Pi Zero W

### Camera

I am using the following type of camera: MMAL

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue: N/A

### Log Files

I consider the following log files relevant to this issue:

[boot.log](https://github.com/ccrisan/motioneyeos/files/6253340/boot.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/6253341/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/6253342/motioneye.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/6253343/dmesg.log)

I'm having some issues with my motioneyeos installed on a Raspberry Pi Zero W. I got everything to work out properly—successfully flashed the OS, configure names, admin access, recording, and can even access from another device via SSH and FTP.

My problem is after a while, which can be minutes or hours, I can't access it anymore. Not via SSH, FTP, or the IP address on browser.

I keep getting a "connection timeout" or "can't connect to server" error.

When i pull the plug and put it back on, it reloads, and now i can access everything again. I'm not sure how to troubleshoot it.

motionEye Version | 0.42.1
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606

Originally (wrongly) posted here: https://github.com/ccrisan/motioneye/issues/2086

starbasessd, 2021-04-03:
Missing a log file. Are you able to attach a keyboard & mouse?
Can you ssh in or get on the console (keyboard and mouse) and run the following commands, and post the results, please?
vcgencmd measure_temp
vcgencmd get_throttled
This will tell us what the Pi is running at temperature-wise, and if there are issues with power.
It would be helpful to see what is going on the console when you can no longer reach the Pi via network.
It could be you have a flaky WiFi chip, or it's locking up due to overheating.
It is also possible, due to the camera attached, it is drawing too much power from your power supply. You could test that by removing the camera and see how long it stays up.

starbasessd, 2021-04-03:
I also note a number of soft shutdown/reboot cycles, is that something you were doing?

arielylim, 2021-04-03:
I added the other log file in the original post. 

I can access it but have to restart again. Let me do that after this. I tried SSH-ing now and im getting a connection timeout. 

What did you mean by this? 
> It would be helpful to see what is going on the console when you can no longer reach the Pi via network.

I have a feeling it's not the camera. Because while i was typing my original post, i disconnected it first, then plugged the pi in front of me. Basically it only has the SD card attached and power supply. That was when I posted this second comment with the image https://github.com/ccrisan/motioneye/issues/2086#issuecomment-812919116

After removing everything, putting in a relatively cool room (don't have a thermometer but i can assume around ~72-ish F). Plugged in, worked for about 30 minutes, then the connection error. 

As for the shutdown/reboot, that could be me unplugging the pi and plugging it back in. Top of my head, that's less than 10x. If it's more, then the other causes aren't me.

arielylim, 2021-04-03:
I am using the charger from canakit pi zero complete starter kit. 2.5A 

I just unplugged/plugged the zero, and here's what i see with the temp. 

[image]

Not sure if this was how it was before since it might have cooled down a bit, if it's the temp that's causing this.

arielylim, 2021-04-03:
Added info: 
- Seeing the temp hovering around 49.8 - 50.8 C
- Peaked at 51.9 C, but after checking again it's the same as above.
- Camera/Pi is running about 5 minutes now. Still working.

starbasessd, 2021-04-03:
"It would be helpful to see what is going on the console when you can no longer reach the Pi via network."
If the monitor is showing the normal logon screen, something corrupted, or something else.
The boot.log is showing a 'normal' shutdown (maybe commanded, maybe something else.
Under Settings, Expert, there is another log: messages.log, that was the 5th one I need.

arielylim, 2021-04-03:
Ah, yes. When i get it to boot using the unplug/plug method, i shut it down using the web browser before repositioning/ 

Original setup is by the window and a plug near. When it stopped working, i unplug it, bring it to my working desk, plug it near there so i can access on my table with my laptop/monitor. Everything is working fine here. I shut it down from the admin browser. Return it back to its spot by the window. 

Reattached all the new log files. 
[boot.log](https://github.com/ccrisan/motioneyeos/files/6253468/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/6253469/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/6253470/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/6253471/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/6253472/motioneye.log)

starbasessd, 2021-04-03:
Are you using the same power brick in both locations?
Did it run continuously overnight 4/2 8p to 4/3 8a?

arielylim, 2021-04-03:
> Are you using the same power brick in both locations? 
Yes. Window is direct plug/outlet while the one near my table is using a UPS

> Did it run continuously overnight 4/2 8p to 4/3 8a?
Yes, that's when i first encountered the issue. I just finisehd the setup last night. Around 8pm. Everything was fine, i was viewing livestream via the ip:8081 on my phone until about midnight. 

Went to sleep and when try checking this morning, it won't load. That's about 7:30-ish. I started the fitst unplug/plug thing around 8am. 

Additional info: 
so after i uploaded the logs, everythign was working for a few minutes. This was already by the window (original location). It hanged / connection error again. Couldn't access via browser and my SSH connection wherei m' checking the temp got disconnected. 

I went and did something else. Just now, upon reloading the browser, i can access it again. SSH and temp is at 51.9C.

starbasessd, 2021-04-03:
Do you have the ability to connect a monitor to it?
Can you re-image with dev20201026? 
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz

starbasessd, 2021-04-03:
Before doing anything else, can you run the 
vcgencmd get_throttled 
again?

arielylim, 2021-04-03:
>  vcgencmd get_throttled

Done. 
[image]

arielylim, 2021-04-03:
I also have monitor and keyboard connected to it right now.

starbasessd, 2021-04-03:
Wait until it doesn't respond to the browser or ssh, then see if the
monitor and keyboard are responding. If they are, then run the 2 vcgencds
again.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

arielylim, 2021-04-03:
the waiting game. will do. 

What if i can't access the pi via the attached keyboard too? 

Should I go ahead with the dev release?

starbasessd, 2021-04-03:
The software/SDCard is failing and crashing the whole Pi vs only the WiFi is failing. 
If the keyboard is working then something about the WiFi/Network is failing.
You can try pinging your router/gateway, and run the 2 vcgencmds again
If the keyboard also fails, then yes, please reflash to the dev20201026

arielylim, 2021-04-04:
I left the pi running and now after trying, i can do the two vcgencmd. But getting a timeout error.

arielylim, 2021-04-04:
[image]
I tried pinging the IP from the router admin (see image) and it seems to work. Don't know if that helps.

arielylim, 2021-04-04:
this info might also help rule out a few things. Since i can access the zero with the keyboard/monitor, i turned it off using `poweroff` command. 

I turned it back on, tried FTP access  from my laptop and i can confirm the camera was still working/recording despite not being accessible via SSH or browser IP. The video of me walking up to my desk, typing the vcgencmd, etc. are recorded. So the motion detection and recording is working. 

From the comment earlier, it could be the wifi failing. 

1. Does that mean motioneyeos will keep recording even without wifi/internet? From what happened, this seemed to be so. I'm currently saving to the onboard SD card, but plan to move to a server/hub setup where other Pi Zeros will save to a hub Pi or a NAS. Will that still work if wifi connection is lost? 
2. if it is indeed the wifi, what do i do? Do i return the zero to the seller? Got this for just a little over 24 hours ago. I got it solely for this project.

starbasessd, 2021-04-04:
I meant ping the router from the Pi. Does outbound work?
If you haven't re-imaged, please run the 2 vcgencmds from the Pi console (monitor & keyboard) when you can't connect to the IP address via ssh or web browser.
In the Web Browser, when it works, turn on:
Settings, Expert Settings, Connectivity Watch, and accept the default settings.
I would also seriously consider re-imaging to the dev20201026 package.

arielylim, 2021-04-04:
> If you haven't re-imaged, please run the 2 vcgencmds from the Pi console (monitor & keyboard) when you can't connect to the IP address via ssh or web browser.

I did. It was working. Temp is around 49c last night. 

I also did the connectivity watch. Still timing out when i tried earlier this morning. 

I'll re-image using the dev. Hopefully it'll work. 

Sidenote: Could it be my network/router? or the wifi of the pi? 

Because it is working/recording. I just pulled it out, plugged it back in then tried FTP. I got several recordings during the time I can't SSH/access via browser.

arielylim, 2021-04-04:
I just installed the dev version and it's working so far. I'll monitor this and report back here. 

To elaborate the FTP earlier (https://github.com/ccrisan/motioneyeos/issues/2714#issuecomment-813085865): 
- Last night, i placed the zero/cam by the window again. As usual, after booting, everything is working fine. I left it to run overnight. 
- This morning, i tried viewing via browser on my phone. I'm getting the connection error. After a while, i tried via laptop. Same thing. Tried SSH, timed out as well.
- Pulled the plug, moved it to my workstation, plugged it in. I can connect to it again. 
- In order to verify the hunch i had mentioned https://github.com/ccrisan/motioneyeos/issues/2714#issuecomment-812965953, i FTP into the PI and confirmed it was recording during the time I can't access it. 

> Could you ftp prior to pulling it out and plugging it back in, or was it down too?

I tried this last time, it was giving me timed out errors as well. 

Based on what everything so far, here's what I learned: 
- After every reboot, whether that's proper shutdown/reboot or unplug/plug, the device is working properly. I can connect to it every single time. 
- After a while, I can't connect anymore. Random--sometimes minutes, sometimes hours (like overnight).
- After the "connect pi with monitor/keyboard" test, the connection (ssh, ftp, browser) still hanged. But keyboard/monitor is working (i.e. can type the vcgencmd and even shutdown via 'poweroff')
- when i can't connect, all three methods (SSH, browser, FTP) aren't working. 
- During the times I can't connect, the pi is still activating recording for motion and recording in the SD card (as confirmed in the message above)

arielylim, 2021-04-05:
decided to try again and take some screenshots. Here, browser and FTP can't connect. 

[image]

But when i plugin the pi to the TV (it's closer to the window) and the keyboard, i can still login and type. 

https://www.dropbox.com/s/70ro13lrqd4pqe0/IMG_0818.mov?dl=0

Do you think it could be a hardware issue? 

I'm using this already:

motionEye Version | 0.42.1
Motion Version | 4.3.1
OS Version | motionEyeOS dev20201026

arielylim, 2021-04-06:
Could it be a wifi issue? I tried running pihole on the same pi (same hardware, same connection, same sd, etc.) and it exhibits the same behavior. After a while, can't ssh/connect to the browser. 

I searched several forums and found that the power management of the pi could be an issue. 

I added this line to the '/etc/rc.local' file and let's see what happens. 

`iwconfig wlan0 power off`

hoping this solves the issue. 

But just in case, will this affect something in motioneye?

starbasessd, 2021-04-06:
If it is, it's really weird, becaue it's a PiZeroW, and the only network is WiFi (unless you have an Ethernet USB adapter) and it auto reboots if it loses network connectivity, and boot.log would show the panics. 
You can't add that to anything in /etc that file system is read-only. You could do it in /data/etc.
Power management shouldn't be an issue, either. 
Do you have issues with any other wifi connected devices?

arielylim, 2021-04-06:
> unless you have an Ethernet USB adapter

No i don't. Only the pi and camera by itself. Then when i need to plug to monitor and use keyboard, i use the OTG adapters, etc. 

> Do you have issues with any other wifi connected devices?

No, just this pi. Laptop, phone, etc don't have issues as well. I have a 4B that doesn't have any issues too, though i'm using it more as a desktop and occassional retropie gaming.

arielylim, 2021-04-27:
After lots of testing, I found the issue. The SD card. Not sure what's wrong but after switching to a 3rd one, timeouts no longer happen. 

I tried a new pi0 with the same card and PSU, same thing. Tried the 2nd card, and same thing. Tried changing the OS on both, same thing. Even tried changing the power supply. 

Then I bought new SD cards and tried those. No problem anymore. Maybe I'm just unlucky to have lucked out on two consecutive sd cards that I needed the 3rd one to isolate the issue. Thanks for the help!

closed 2021-04-27 by arielylim

gasimoff, 2024-10-16:
@arielylim, it is too late maybe but can you please write which card caused the issue and which one worked fine?