motioneye-project/motioneyeos #2715: Can I take a picture at 12:00 every day?

Jonatan-IG, 2021-04-05:
Can I take a picture at 12:00 every day?
I wanna create a timelapse of snow melting, I am running Motion Eye in Home Assistant OS.

starbasessd, 2021-04-05:
This is more appropriately a question to ask in the motionEye issues github here:
https://github.com/ccrisan/motioneye/issues
Not as a built in function. You could run a cron tab job calling the web trigger for Snapshot, theoretically. I have not seen any scripts like that, though.
You could set the camera up a second time (supposing you wanted to use it normally, too) and select Still Images, Capture Mode Interval Snapshots with an interval of 86400 seconds, but there isn't a way to select a specific time, so you'd have to start it at midnight manually.

Jonatan-IG, 2021-04-06:
Can I in any way set my FPS to one frame per minute, AFAIK I can only set 0.1 fps (on frame per 10 seconds)

starbasessd, 2021-04-06:
You could set the camera up a second time (supposing you wanted to use it normally, too) and select Still Images, Capture Mode Interval Snapshots with an interval of 60 seconds. One of the features of viewing is to create a Time lapse of the images (at the bottom of the Viewer)

snoopy0815, 2021-04-11:
You could add a script to crontab - stopping MotionEye, take a picture - and restart MotionEye again.
This is how I did it:

- open bash with ssh at your motionEye device
- get write-access:
     mount -o remount,rw /
- open new file:
      nano /usr/sbin/take_cyclic_pic.sh
- copy this script into it:

`
#!/bin/sh
pfad=/data
/etc/init.d/S85motioneye stop
raspistill -q 75 -ae 32,0x00,0x808000 -a 4 -a "Terrasse %Y-%m-%d %X" -o $pfad/"$(date +"%Y%m%d_%H-%M-%S")"_Terrasse.jpg
/etc/init.d/S85motioneye start
`

- save file (ctrl-o  and then ctrl-x)
- make file executable:
      chmod 777 /usr/sbin/take_cyclic_pic.sh
- add it to crontab in order to be executed every day at 12 o clock:
- open editor
      nano crontab -e

- add these line:
     0 12 * * * /usr/sbin/take_cyclic_pic.sh >/dev/null 2>&1 &                 take_cyclic_pic.sh   every day at 12 oclock
- save by:   Ctrl+O   /   Ctrl+X

ready...  :)

cheers, j:)rgen.

--------------------

further options for raspistill - the program which takes the pictures:

`options for raspistill:
-vF		flip vertically
-vH 		flip horizontally
-o xyz 	filename
-q 75	jpeg quality 75%  - not used here
-ae		annotate text white   (-ae 32,0xff,0x808000     --> gives size 32 white text on black background)   z.b. A0A0A0  für grau
Specifies annotation size, text colour, and background colour. Colours are in hex YUV format.
Size ranges from 6 - 160; default is 32. Asking for an invalid size should give you the default.
-a		4 - time - and text and date
`