motioneye-project/motioneyeos #2716: OSError: [Errno 2] No such file or directory: '/etc/motioneye'

mf76130, 2021-04-11:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20200606

### Board Model

I am using the following board/model: Pi 3B+

### Camera

I am using the following type of camera: Network Camera

My camera model is: Wansview, Maginon IPC-100HD

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/6291496/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/6291497/motion.log)

starbasessd, 2021-04-11:
Is this just a report on the error, or is it affecting the system?

mf76130, 2021-04-11:
Its annerror in the log, the system works, but maybe getting rid of that might improve performance?