motioneye-project/motioneyeos #2717: [motioneye]  WARNING: mjpg client connection for camera 1 on port 8081 is closed

mf76130, 2021-04-11:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20200606

### Board Model

I am using the following board/model: Pi 3B+

### Camera

I am using the following type of camera: Network Camera

My camera model is: Wansview, Maginon IPC-100HD

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/6291501/motioneye.log)

starbasessd, 2021-04-11:
Is this just a report on the error, or is it affecting the system?

mf76130, 2021-04-11:
Its just in the logs, butbi hope i can improve performance of the system in case i get rid of it?

starbasessd, 2021-05-28:
What kind of performance are you getting, and what do you expect?
What kind of image resolution are you using?
What framerate are you attempting?
Is the camera Ethernet or Wifi?

mf76130, 2021-05-28:
I have 9 Cameras, most of them are on 480p and only 3 are on 720P, Frame rates are between 5FPS and 13FPS depending on the wifi reception, so all cams are on wifi.
The performance is Okish, its not perfect but kind of working.
So when recording starts the framerate sometimes drops and the connection gets bad.

starbasessd, 2021-05-28:
Sounds like you might be saturating the wifi network. Is it 802.11 A, B, G, N, AC, or AX?

mf76130, 2021-05-28:
i have unifi uap-ac-lite, so the bottlenecks are the cameras :-) as those are only NG, and some only get -50dBm

starbasessd, 2021-05-28:
Still depends. If your ethernet network can't support gigabit to the AP... What is your motionEye hub? If it is a Pi3B, it doesn't have gigabit, It only has 100Mbit ethernet, and could be saturated at that end. 3B+ and 4s have Gigabit (sorta) and you get at least 300Mbit That's a lot of cameras. I'm not sure how else to help. Another AP might help with signal strength. Switching your hub to a refurb TFF desktop / X86 (like my Lenovo M73) might be a better choice for you. My RPi4-4GB can handle 6 cameras (3xd-Link @ 640x480, 2xFosCam @ 1280x720, and 1 Generic TIGA dashcam @1024x768 without too many issues.

mf76130, 2021-05-29:
everything is gigbait in my house, only Pi3B+ not.
using Top i see pi never going above 70% on the NIC.
I tried Pi4 when it was released, but at that point in time i had more problems running it stable and i didn't see any NIC issues.
But i assume you are correct, i am on the limit with 9 cams :-)
I have some M73 at home as well, but the PI uses less space in the server rack und less power :-)
Thanks for you analysis and hints