motioneye-project/motioneyeos #2718: Multiple Cameras Not Working and Other USB Issues

rob585-538, 2021-04-12:
### motionEyeOS Version
Version 20200606

### Board Model
Raspberry Pi 4 Model B 2GB Quad Core 64 Bit Cortex-A

### Camera

I am using the following type of camera:  V4L2 (this is the option I select (as its the one one available). It is a camera that goes directly onto the  motherboard

**My camera model is:**
MakerHawk Raspberry Pi Camera IR Fisheye Wide-angle 150-160 Degree 5MP OV5647 Webcam Automatically Switching between Day-Vision and Night-Vision Shooing Mode for Raspberry Pi 2B 2B+ 3B 3B+ 4B

**### Network Connection**

Wifi (but also tried with Ethernet)
### **Peripherals**

SD Card for O/S (on the motherboard)
I have a mouse and keyboard attached - to the USB but they don't respond - e.g I can't log in locally (I don't need to, but just checking them)
I have tried connecting a Logitech USB web camera that doesn't work. 

**Issue**:
I cannot see any pictures/video with the error "Unable to Open Video Device". The camera works - I know this because I have a separate SD card with a raspberry image when I switch this on I know it works. 

MotionEye does not appear to recognize this camera properly, or the Logitech web cam.  I am open to  options and could buy a new camera - I just don't know which one to buy. 

[image]
[image]

starbasessd, 2021-04-12:
The type of camera you have, connecting with a ribbon cable, is MMAL, and should be auto-discovered. There are issues with release 20200606 and Pi4s. Please try dev20201026.

rob585-538, 2021-04-16:
Thank you for this - and sorry the delay in responding. I have since got the updated dev20201026. and this fixed everything . Brilliant support from ccrisan as well.  Thanks again

closed 2021-04-16 by rob585-538