motioneye-project/motioneyeos #2721: MMAL Camera not detected by rpi Zero W

nichtAlex, 2021-04-16:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
I am running motionEyeOS version: 20200606

### Board Model
I am using the following board/model: Raspberry Pi Zero W

### Camera
I am using the following type of camera: MMAL (I think. It's an IR Camera Module for the Zero W from AliExpress)

### Network Connection
My motionEyeOS unit is connected to the network via: WiFi
 
So i've managed to set up MotionEyeOs and all and I can access the web interface via the IP. 
When I use PuTTy to ssh into the Zero W and execute "vcgencmd get_camera" i get "supported=1 detected=0" as an output. The IR LEDs on my camera are lighting up so it definitely gets power. 
I've searched for a couple hours now and tried different things but I just cant get it to work.
What can I try to fix this problem? driver issue? any way i can confirm that its not the camera thats broken?

starbasessd, 2021-04-16:
You can try using raspistill in your ssh session:

raspistill -o cam.jpg

You can try using a different version of motionEyeOS (dev20201026 is
very stable)

You can install raspberryPiOS and try the raspicam commands:
https://www.raspberrypi.org/documentation/usage/camera/raspicam/README.md

In all the above, though, it may be a bad cable (try re-seating the
cable at both ends)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

nichtAlex, 2021-04-17:
OK so I tried everything you mentioned except installing raspberryPiOS as i dont have a keyboard or mouse for the Pi Zero. 
Ive also reseated the cable several times. 
executing raspistill -o cam.jpg gives me this error:
`[root@meye-afe68cce ~]# raspistill -o cam.jpg
mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly`

starbasessd, 2021-04-17:
You can install RPiOS the same way you install motionEyeOS (same
wpa_supplicant.conf and ssh files) to make it 'headless'
If you then connect, run raspi-config, enable Interfaces, turn camera on.
Then run the same commands.
Otherwise, looks like a broken camera, or unsupported one. You can also try
motionEyeOS version dev20201026.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

nichtAlex, 2021-04-17:
for some reason the camera is now being detected and the video stream also appears in the web interface. 
all i did was remove and reattach the camera lens which is only held down by two screws.

starbasessd, 2021-04-17:
Probably just a loose connection. Not knowing the specifics of that camera,
I'm sorry I didn't have you check it before.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

nichtAlex, 2021-04-17:
Anyways Im thankfull for all replies I got.

closed 2021-04-17 by nichtAlex