motioneye-project/motioneyeos #2724: still image - motion triggered with snapshot interval

nopub, 2021-04-19:
Hi,

I’m running successfully motioneyeOS on a Raspberry Zero W with a raspberry cam.

motioneyeos 0.41
Motion Version 4.2.2
OS Version motionEyeOS 20190911

I know it is not the latest version but setup is currently monitoring a occupied bird nest inside a box through a  private intranet network and do not want to change the current settings or rebaseline the sd card.

I have two questions/requests/enhancements (potentially present on a more recent version).

	- for still image configuration : would it be possible to take pictures when 			motion triggered (existing feature) but with a predefined snapshot interval, for a timelapse 	movie?

	- when you go to the list of movies recorded via the web interface, would it be possible to 	have button to download all movies at once (the same way you have for delete all).

I have anyway a problem accessing motioneyeos files thru Nemo on Mint.
Through smb I can see sdcard and storage. While I’m’ able to see the files under storage, I cannot access the files under sdcard. Credentials are not recognized. I tried admin, root with no password, I tried workgroup WORKGROUP and workgroup MOTIONEYEOS withoiut success. 

Any idea. I have no problem with ssh and sftp

Thanks,

Fabrizio

starbasessd, 2021-04-19:
I agree with the single bulk download (and delete) feature in the WebGUI would be 'nice', and the snapshot interval as a motion detected trigger has been requested before.
As to the access for [ip_address]/sdcard vs [ip_address]/storage:
Works in Windows File Manager (prompts for credentials and lets me in)
Checking in several Linuxs, will report shortly.

starbasessd, 2021-04-19:
Linux in general doesn't like 'no password set' connections, at least to motionEyeOS. Nautilus and Nemo tested in several Debian flavors (Mint is Ubuntu/Debian) and all had the same issue. Windows 10 did not have an issue. After I set a password (admin:admin, and user:user) both were able to access with no further issues to both /sdcard and /storage. 
Domain was left at 'WORKGROUP' when testing.

nopub, 2021-04-20:
Thanks for th feedback.

ubuntu748, 2021-04-22:
motioneyeos on the RPi comes with raspivd and raspistill - so you should be able to run a script that takes a snapshot at any specified time.