motioneye-project/motioneyeos #2726: "Enable Samba Server" not working

J-o-h-n-M, 2021-04-20:
### Preliminary Docs
I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
I am running motionEyeOS version: motionEyeOS dev20201026

### Board Model
I am using the following board/model: Raspberry pi 4

### Network Connection
My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals
I am using the following peripherals that I consider relevant to this issue: External harddisk connected to USB 3

I am trying to connect to the USB disk that I have connected to my RPI4 where MotionEyeOS is running. But cant seem to be able to connect from a window machine. It wont show up in network discovery. But also typing the hostname or IP address results in nothing.
I have also changed the smb.conf to use workgroup "WORKGROUP" but that does not help.
If I use the "Enable FTP Server", I can connect with FTP.

Any hints/suggestions would be greatly appreciated.

my smb.conf (after modifying the workgroup only):
```
 [global]
   workgroup = WORKGROUP
   server string = motionEye
   security = user
   map to guest = bad user
   encrypt passwords = yes
   private dir = /var/lib/samba/private
   public = no
   writable = no
   include = /data/etc/smb.conf
   load printers = no
   printing = bsd
   printcap name = /dev/null
   disable spoolss = yes
   log level = 0
   syslog = 0
   preferred master = no
   domain master = no
   local master = no
   os level = 0

[sdcard]
   comment = SD Card Output Directory
   path = /data/output

[storage]
   comment = Attached Storage Devices
   path = /data/media
```

starbasessd, 2021-04-20:
What changes, if any, did you make to the WebGUI?
Which Pi4? 1, 2, 4, or 8 GB?
How is your USB drive formatted?

J-o-h-n-M, 2021-04-20:
I just have "Enable Samba server" to ON 
and "Require Samba Authentication" to OFF

Pi4 4GB

I dont know how the drive is formatted. Its a brand new WD digital external USB3 drive. But files are being written to it, as I can see them when I access it with FTP

starbasessd, 2021-04-20:
I just set up a Pi4-4GB with motionEyeOS dev20201026
Set up a camera (network)
Turned on Samba write and FTP write in Services and rebooted as prompted
In a Win10 machine, opened File Explorer
Typed in \\[IP_Address]\storage
It popped up a login screen, put in admin, nothing for password, and it let me straight in.
Ditto for \\[IP_Address]\sdcard
popped up a loging screen, put in admin and nothing for password, and straight in.
No need to modify smb.conf

J-o-h-n-M, 2021-04-20:
weird, perhaps I just need to reinstall then. But basically I did the same thing. One thing I noticed is the line "SMB1 disabled -- no workgroup available" in the command below. Not sure if that matters

```
[root@Motioneye ~]# smbclient -L motioneye
Enter WORKGROUP\root's password:
dos charset 'CP850' unavailable - using ASCII

        Sharename       Type      Comment
        ---------       ----      -------
        sdcard          Disk      SD Card Output Directory
        storage         Disk      Attached Storage Devices
        IPC$            IPC       IPC Service (Samba 4.11.10)
SMB1 disabled -- no workgroup available

```

starbasessd, 2021-04-20:
You're making a lot of changes, some not even in the GUI.
My output
-----
[root@meye-2ba2d181 ~]# smbclient -L meye-2ba2d181
WARNING: The "encrypt passwords" option is deprecated
WARNING: The "syslog" option is deprecated
Enter MOTIONEYE\root's password:
dos charset 'CP850' unavailable - using ASCII

        Sharename       Type      Comment
        ---------       ----      -------
        sdcard          Disk      SD Card Output Directory
        storage         Disk      Attached Storage Devices
        IPC$            IPC       IPC Service (motionEye)
SMB1 disabled -- no workgroup available
-----
You have to realize this is a highly stripped down, heavily modified system, using buildroot and thingos, not a normal version like RaspberryPiOS or Ubuntu. It doesn't always act like a normal Linux.

J-o-h-n-M, 2021-04-20:
Well this is interesting. I did a full reinstall of MotioneyeOS, with the exact same results.

I then went to an older windows notebook and typed \[IP_Address]\storage  and it works there!

So it seems to be something with my newer Windows notebook then and not a motioneyeos issue.

closed 2021-04-20 by J-o-h-n-M

J-o-h-n-M, 2021-04-20:
Anyone reading this issue later. The solution for me was on this page https://devanswers.co/network-error-problem-windows-cannot-access-hostname-samba/

starbasessd, 2021-04-20:
Might also be a bonjur issue (avahi/mDNS) difference between your Win10 machines. I have bonjur installed as part of my base setup (I also have a MacOS machine on my network, along with all the other hosts and VMs and iPhones and Android phones...