motioneye-project/motioneyeos #2727: Additional camera driver

powerfischiMF, 2021-04-22:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20200606

### Board Model

I am using the following board/model: Raspberry Pi 4

### Camera

I am using the following type of camera: V4L2, MMAL

My camera model is: Veye-MIPI-327

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

Hi, I would like to use a special camera. But i need some additional drivers to be installed. how can I do that? 

I would like to use the Veye-MIPI-327 because its possible to be used in ultra low light conditions.

starbasessd, 2021-04-22:
Due to the modifications and applications they run, you will probably need to install RaspberryPiOS Lite, then motion/motionEye following these instructions:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
Then you can follow the instructions here to install their software:
http://wiki.veye.cc/index.php/V4L2_mode_for_Raspberry_Pi

starbasessd, 2021-05-28:
No response

closed 2021-05-28 by starbasessd