motioneye-project/motioneyeos #2728: Manually copy movies from SD card to USB

wohlfman, 2021-04-22:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2021-04-22:
cp /home/ftp/sdcar/Camera1 /home/ftp/storage/sdax (x being the partition # on the USB

wohlfman, 2021-04-22:
Hello,

Is there a way to copy large movies off of the SD card onto a USB drive?

I am running a raspberry pi model 3b+ with motionEye version 0.42.1, Motion Version 4.2.2 and my OS version is motionEyeOS20200606. When I initially set this up I went over board and created some huge movies that I am unable to watch or download over my network. 

I have since scaled back to a more manageable (default) video size and can both watch and download videos. Currently all movies are stored on the SD card and I am ok with that however I want to collect the large movies to make additional room. 

Thanks in advance!

starbasessd, 2021-04-22:
cp /home/ftp/sdcard/Camera1 /home/ftp/storage/sdax (x being the partition # on the USB
or
mv /home/ftp/sdcard/Camera1 /home/ftp/storage/sdax (x being the partition # on the USB

wohlfman, 2021-04-22:
Thanks for the speedy response starbasessd. I will give this a shot when i am able to get the camera down.

starbasessd, 2021-04-22:
You connect to the SSH on the camera on the network, then run either of
those commands, assuming you have a USB drive already plugged in.
If not, you can SMB (Windows Share) or FTP download the files, then use the
SSH terminal to delete the files.
If you'd like to try that, let me know.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-05-28:
Status?

closed 2021-05-28 by starbasessd