motioneye-project/motioneyeos #2729: Possible to Have Hub Also Display Webpage? ::Question::

zalterego36, 2021-04-26:
Is it possible to have a single, more powerful Raspberry Pi (a 4b for instance) serve as both a hub and a display for the webpage that collects all the cameras together?

This is such a wonderful project, by the way. Thank you so much. I'm a teacher and I've recently been put in charge of four separate spaces in my school and it's so useful to have extra eyes on the local network. I'm only wondering if the display thing is possible because I'm not ALWAYS the supervisor, so for the times another teacher is there it would be easier to just have a monitor that's always running the webpage and I figured the Pi4 I'm using as a hub can do extra duty (I hope).

Thank you for answering and I appreciate all you're doing.

starbasessd, 2021-04-26:
The hub would need to be motioneye on RaspberryPiOS instead of motionEyeOS, but it is doable...

zalterego36, 2021-04-26:
Huh. Didn't even think of that. Thank you, I'll take a look at what I can put together.

closed 2021-04-26 by zalterego36